mysensors/MySensors #1160: Built-in handling of watchdog timer

Avamander, 2018-07-05:
It would be nice if MySensors had built-in handling automatic of AVR watchdog timers. So that the user wouldn't have to deal with initialization, and resetting when using `sleep()`, `send()` and etc. Preferably this could be enabled with just one preprocessor flag. 

Having the node let the controller know the device has encountered a fault would also be nice but that would require a new data type and explicit controller support, right?

flatsiedatsie, 2018-12-05:
Could you elaborate on what MySensors does to the watchdog? After which commands is the watchdog disabled? And.. why does MySensors do this?

mfalkvidd, 2018-12-05:
Ongoing discussion in https://forum.mysensors.org/topic/9946/watchdog-not-watchdogging/