mysensors/MySensors #1161: System data type for reporting errors

Avamander, 2018-07-16:
It seems that currently there seems to be no way to report sensor and actuator errors to controllers, this could be remedied by creating an internal data type for reporting sensor errors (one of the values could be replaced maybe?). This would make MySensors much much more reliable because then users will actually know when something's erroring rather than not getting sensor readings for example.

mfalkvidd, 2018-07-16:
How about using I_LOG_MESSAGE or I_DEBUG ?

Avamander, 2018-07-16:
@mfalkvidd It would work assuming gateways understood the format, what I was thinking would be more like `V_ERROR` with certain values indicating certain types of errors that could be hardcoded into gateways and thus nicely displayed to users. Currently every single person wanting this functionality has to personally implement both the gateway part and how (s)he's going to send the errors.

mfalkvidd, 2018-07-17:
Gateways don't display anything to users (except the debug log, where I_LOG_MESSAGE and I_DEBUG already are visible).

Do you mean that you wish controllers implemented a way of displaying these types of messages?

Avamander, 2018-07-18:
@mfalkvidd Sorry, I indeed meant controllers.