mysensors/MySensors #1171: Feature request: implement PJON protocol for wired communication

flatsiedatsie, 2018-08-05:
Wired communication is more secure and private by default. This protocol looks great.

https://forum.mysensors.org/topic/3977/transport-based-on-single-wire-pjon-protocol/17

flatsiedatsie, 2018-08-18:
@gryzli133 Do you mean that it can already be used as the hardware layer right now? Without requiring any change in MySensors?

Looking at the protocol further, the promise of "just connect a wire" isn't that simple.
https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang

- "A 1-5 MΩ pull down resistor could be necessary to reduce interference"
- "Pins can be optionally protected against overload adding a current limiting resistor to each connected pin"
- And of course you will in most cases also need an additional ground wire to create a common ground.

gryzli133, 2018-08-18:
@flatsiedatsie unfortunately I dont think this is included already... this is why it would be perfect to include it as new transport layer, because it really just need one wire. I'm testing it on my table for quite long time and it looks very reliable. The user 4ib3r has write to me, that his custom library for PJON is working well on MySensors by him since 2 years now

jwaggy, 2019-03-05:
I agree that PJON would be a great addition.

M-m-X, 2019-03-31:
Yes, a lot of wireless protocols and very little wired protocols.
In addition if it is very reliable, great!

PR #1278 (2021-03-20): Add PJON transport layer
- Based on #1230, credit to @4ib3r
- PJON library updated to 12.1
- Fixes #1171
closed 2021-03-20 by tekka007