mysensors/MySensors #1172: Feature request: simple security on the gateway

flatsiedatsie, 2018-08-05:
It would be great if setting up security on the Raspberry Pi side of things could mirror the simple option in the nodes.

The [current situation](https://www.mysensors.org/apidocs/group__SecuritySettingGrpPub.html#gac90f5b712d34ad4a56354cccee8857c4) is a little confusing and counter intuitive:

- the gateway counterpart to MY_ENCRYPTION_SIMPLE_PASSWD is -my-security-password.
- the counterpart to MY_SIGNING_SIMPLE_PASSWORD is --my-signing=password --my-security-password
- the counterpart to MY_SECURITY_SIMPLE_PASSWD does not exist.

It would be great is the commands for the gateway were the same as the commands in the nodes.
So for example, all that would be required to secure the network on the gateway side was to add --MY_SECURITY_SIMPLE_PASSWD="123456789"

mfalkvidd, 2018-08-06:
I am not familiar with the signing stuff, but a potential workaround could be to just add MY_SECURITY_SIMPLE_PASSWD using the instructions provided at https://www.mysensors.org/build/raspberry#advanced

flatsiedatsie, 2018-08-08:
I don't quite follow. That command not being available is the whole point of my feature request :-)

I also don't think the --extra-cxxflags="-MY_SECURITY_SIMPLE_PASSWD" could work. I suspect he flag doesn't exist yet in the gateway code, and this method doesn't offer any way to set the password.

flatsiedatsie, 2018-08-09:
I've had a look, but it's over my head I'm afraid. And I wouldn't even know which command the new flags would need to encapsulate if I'm honest.

flatsiedatsie, 2019-04-19:
This would still be so very cool to have.

fallberg, 2019-04-30:
Sorry, I do not have the bandwidth to look into or validate this right now. But it should just be a matter of updating the configure script. There is already an argument for providing passwords for both signing and encryption, so just add a new argument that then provide the other two arguments internally and you should get what you want.