mysensors/MySensors #1195: Orange pi zero + Armbian Stretch + nrf24 does not work

Haba1234, 2018-08-30:
Hi!
 I configured DT overlays by adding (armbianEnv.txt):
`verbosity=1`
`logo=disabled`
`console=both`
`disp_mode=1920x1080p60`
`overlay_prefix=sun8i-h3`
`overlays=spi-spidev spi-add-cs1 uart1 usbhost2 usbhost3 w1-gpio`
`param_spidev_spi_bus=0`
`param_spidev_spi_cs=0`
 I had spidev working, but not the mysgw. It doidn't give me errors, but i was stuck at !TSM:INIT:TSP FAIL.
Any ideas?

mfalkvidd, 2018-08-30:
Several orange pi users have reported similar problems in https://forum.mysensors.org/post/92885 but so far it seems like no-one using orange pi has been able to figure out a solution.

marceloaqno, 2018-08-31:
I did a test with the the [nRF24/RF24](https://github.com/nRF24/RF24) library and the RF24 module was successfully detected in Armbian Stretch (4.14.65-sunxi):
```
pi@orangepizero:/tmp/RF24/examples_linux$ sudo ./gettingstarted
RF24/examples/GettingStarted/
STATUS           = 0xe7 RX_DR=1 TX_DS=1 MAX_RT=0 RX_P_NO=3 TX_FULL=1
RX_ADDR_P0-1     = 0xcfcfcfcfcf 0x9f9f9f9f9f
RX_ADDR_P2-5     = 0x3f 0x3f 0x7e 0x7e
TX_ADDR          = 0xfcfcfcfcfc
RX_PW_P0-6       = 0xf9 0xf9 0xf3 0xf3 0xf3 0xe7
EN_AA            = 0xcf
EN_RXADDR        = 0x9f
RF_CH            = 0x3f
RF_SETUP         = 0x7e
CONFIG           = 0xfc
DYNPD/FEATURE    = 0xf9 0xf9
Data Rate        = 250KBPS
Model            = nRF24L01
CRC Length       = 16 bits
PA Power         = PA_MAX
```

I'll check what needs to be changed in MySensors to make it work.

Haba1234, 2018-08-31:
Thanks, I will wait for the results.
Orange pi zero warms less on Armbian Stretch

closed 2018-09-04 by tekka007

Shaggy013, 2018-09-08:
it is just this patch on the bottom on that page what did the fix for me on  opi one  ,pc plus , pc2 and zero plus2 h5

http://linux-sunxi.org/SPIdev

 HIGH on SCK line right before transfer

Starting an SPI transfer via sun6i-spi and sun4i-spi might raise the SCK line to HIGH shortly before transfer. Such behaviour might confuse slaves, especially when not using / ignoring chip-select.

This is due to SUN6I_GBL_CTL_BUS_ENABLE being written into SUN6I_GBL_CTL_REG at an early stage. Moving that to the transfer function, hence, right before the transfer starts, mitigates that problem.

Following patch does so for spi-sun6i:

diff --git a/drivers/spi/spi-sun6i.c b/drivers/spi/spi-sun6i.c
index 8533f4edd00a..6a14589cce32 100644
--- a/drivers/spi/spi-sun6i.c
+++ b/drivers/spi/spi-sun6i.c
@@ -304,6 +304,9 @@ static int sun6i_spi_transfer_one(struct spi_master *master,
 
        sun6i_spi_write(sspi, SUN6I_CLK_CTL_REG, reg);
 
+       /* Finally enable the bus - doing so before might raise SCK to HIGH */
+       sun6i_spi_write(sspi, SUN6I_GBL_CTL_REG, sun6i_spi_read(sspi, SUN6I_GBL_CTL_REG) | SUN6I_GBL_CTL_BUS_ENABLE);
+
        /* Setup the transfer now... */
        if (sspi->tx_buf)
                tx_len = tfr->len;
@@ -411,7 +414,7 @@ static int sun6i_spi_runtime_resume(struct device *dev)
        }
 
        sun6i_spi_write(sspi, SUN6I_GBL_CTL_REG,
-                       SUN6I_GBL_CTL_BUS_ENABLE | SUN6I_GBL_CTL_MASTER | SUN6I_GBL_CTL_TP);
+                       SUN6I_GBL_CTL_MASTER | SUN6I_GBL_CTL_TP);
 
        return 0;

PR #1200 (2018-09-04): Fix SPIDEV on Armbian Stretch
Fix #1195
commit 1c336095 (2018-09-04): Fix SPIDEV on Armbian Stretch (#1200)

Fix #1195