mysensors/MySensors #1204: Add support for Factory reset - I_FACTORY_RESET

jkandasa, 2018-09-07:
At this moment I do not see any option to do a factory reset(erase EEPROM completely) via default library.

It may be a good idea to implement `I_FACTORY_RESET` to perform complete EEPROM erase. This will help us to perform a factory reset from our favorite controller(without toucing the hardware).

As MySensors library supports more than one hardware(ATMEGA328P, ESP8266, etc.,) this feature should be implemented for each platform. End user simply trigger `I_FACTORY_RESET`, when there is a need.

fallberg, 2018-09-08:
This might be a good idea for systems that don't use encryption. Currently all encryption solutions rely on EEPROM for key storage. Distributing a reset command to a node would then effectively exclude that node from future communications. So if this is implemented, care must be taken for users that rely on encryption.

jkandasa, 2019-01-04:
@nekitoss Thank you! Yes, I want to [clear EEPROM](https://github.com/mysensors/MySensors/blob/development/examples/ClearEepromConfig/ClearEepromConfig.ino) in all cases. Not only with MYSBootloader.

flatsiedatsie, 2019-06-08:
> This might be a good idea for systems that don't use encryption. Currently all encryption solutions rely on EEPROM for key storage. Distributing a reset command to a node would then effectively [...]

You could also say that users who have unsecured networks are just as much at risk, since it becomes very easy for a neighbor or war-driver to reset the nodes.

fallberg, 2019-06-17:
Indeed. Just another reason for people to secure their networks :)