mysensors/MySensors #1207: SoftwareSerial started after splash screen

nelsonov, 2018-10-08:
displaySplashScreen is called before the sketch's setup(). If someone is using a soft serial port, that port will not be started until after the splash screen is displayed. This makes the splash screen disappear into a black hole.

Could the call to the sketch's setup() be moved up? Could the splash screen be displayed after setup() where it is now?

mfalkvidd, 2018-10-08:
Can’t the soft serial port be initialized in preHwInit() or before() instead?

tbowmo, 2018-10-08:
It's up to the end user to initialize the softSerial device, as the library cannot guess how the hardware is lined up.

It can be done in preHwInit(), but this is a user task then. Perhaps a note somewhere in the project, that IF you are using softSerial, then you must initialize it by yourself, and do it in preHwInit() function, otherwise you will not see the boot splash.

(at least, that is how I see it 😄)

nelsonov, 2018-10-09:
I wasn't aware that preHwInit existed as a user function.  Putting the software serial initialization in preHwInit does work. I just tested it.  before() also comes after displaySplashScreen(), so preHwInit is the only place it can go.

My intention wasn't for the library to do the initializing the softSerial.  That has to be a user task.  I just wasn't aware the preHwInit() was available.

closed 2018-10-09 by nelsonov