mysensors/MySensors #1212: Feature request: simple MQTT

flatsiedatsie, 2018-10-10:
Currently the MQTT messages that MySensors generates is quite complex. The MQTT path has a lot of details that I don't really need.

Perhaps MySensors could have a simpler MQTT output mode which makes it easier to import/connect in the various controllers? Something like device/sensor/value?

davidgraeff, 2018-10-19:
Actually it would be pretty neat if the Homie convention would be followed. OpenHab 2.4 would be able to autodiscover MySensors sensors.

guillebot, 2018-12-18:
> Actually it would be pretty neat if the Homie convention could be followed. OpenHab 2.4 would be able to autodiscover MySensors sensors.

Yes! I'm done migrating OpenHAB to the new mqtt binding and I completely agree this is the way to go.

I think this could be done all in the mqtt gateway, even working in paralell with all the actual stuff.

Do you (community) think that this is an effort worth doing? Are there any other interested parties?

Please let me know, we could start tinkering with a fork of the gateway.

Regards

tobof, 2018-12-19:
@guillebot Definatly a approch worth some investigation. This might nearly (ID requests need to be handled) make the MQTT part of my MySensors binding obsolete.

guillebot, 2018-12-19:
Yes, probably the MQTT part, but the logic behind the discovery (which is
the most important) not.

Thanks!

tobof, 2018-12-19:
@guillebot Discovery of homie devices / things is supported through the generic MQTT binding:

> [The MQTT Homie convention version 3.x is supported by this binding. It allows to automatically discover devices that follow the "Homie" convention and present them as Things.](https://github.com/eclipse/smarthome/blob/master/extensions/binding/org.eclipse.smarthome.binding.mqtt.generic/README.md)

guillebot, 2018-12-19:
Yes, but mysensors doesn't adhere to homie convention.

So we have the following paths as I see it:

1) Modify mysensors mqtt gateway in order to adhere to homie convention
2) Convince the guy writing the mqtt binding that mysensors protocol is indeed open enough so he will add/allow the discovery of mysensors-convention
2.1.) Fork thay same binding and do it outside Eclipse Smarthome, as many of the non official bindings.
3) Continue to add things manually.  This is not **that** bad, it's just a shame thay having a nice documented protocol in mysensors, we have to add them by hand every time. I'm convinced thay the mysensors platform is one of the bests in terms of software quality, and I'm sure that OpenHAB is indee the best. Just struggling to make both work together.

Thanks a lot for your time

davidgraeff, 2018-12-19:
I'm open to a PR that adds mysensors to the MQTT binding. I will just not do it myself, because I'm a bit in a conflict (biased), because I'm also an author of the Homie convention.

guillebot, 2018-12-19:
Good to know. And thankful.

To be clear, I love the homie convention, and for me the ideal scenario would be for mysensors to add a homie-mode. The whole iot/domotic world need standards more than anyhing else.

As I am a poor programmer both in java and arduino, (Electronic Engineer guy here), I will try to decide which way to explore.

davidgraeff, 2018-12-19:
@tekka007 @mfalkvidd Would you accept a change of the mysensors MQTT layout?

guillebot, 2018-12-19:
> @tekka007 @mfalkvidd Would you accept a change of the mysensors MQTT layout?

Please guys, at least consider it.

The actual MQTT protocol works, is good, and I'm using it, and I'm thankful for all your effort, but is really a 1:1 map of the serial protocol, which is overkill and innecesary for mqtt iot devices.

Thanks to you all.

mfalkvidd, 2018-12-19:
Since a change would break all existing systems, such a change would need to be announced very well in advance, and would require a major release of MySensors (3.0 or 4.0). We should probably reach out to all controller projects that currently support MySensors over MQTT and encourage them to support he new format, and/or help them adapt their code. According to https://docs.google.com/spreadsheets/d/1xQ_654MkTcVmEKyCtcyLd0Tpw8GXRL1fvrUqstN_uNM/edit#gid=471683970 there are currently 8 controllers supporting MQTT but that might not be a complete list.

mfalkvidd, 2018-12-19:
I am not familiar with the mqtt code, but would it be a big difference in development and maintenance to support both the existing format and the homie format and let users selct format by setting a define?

davidgraeff, 2018-12-19:
I agree that is the best solution to allow both.

The file in question is: https://github.com/mysensors/MySensors/blob/development/core/MyGatewayTransportMQTTClient.cpp

Within that file in `reconnectMQTT` a few static publishes (homie version, name, mySensor node) would need to be done. A "last will" need to be established here as well to handle a disconnect properly.

And for every sensor/log/whatever change in `gatewayTransportSend` a publish need to done to the corresponding homie topic as well. No need for any receiving part.

guillebot, 2018-12-19:
Why no need for receiving part? mySensors has both sensors and actuators.

davidgraeff, 2018-12-19:
It's called my**sensors** :D I didn't know the framework can handle actuators as well.

guillebot, 2018-12-19:
Yes. great platform, confusing name.

mfalkvidd, 2018-12-20:
I just noticed tha Jeedom has mqtt support and Jeedom isn't included in the google spreadsheet.

daviddawson, 2021-01-16:
Hello, in case anyone is watching this and might be interested.   I've taken a different route to trying to build in support for either openhab/ homie or anything else into any of the others and constructed a currently very simple MySensors controller that acts as a gateway to Homie.  (in nodejs/ typescript)

It advertises all the MySensors nodes in the network as Homie devices and provides bidirectional comms between the two systems. I've built and tested this using the Homie support in OpenHab 3, and it works well, for where I've got to. 

https://github.com/daviddawson/mysensors-homie-bridge

I'll be expanding it out as I add more to my mysensors setup, all I have at the moment is AC relay control.