mysensors/MySensors #1231: Support stm32arduino

spectrespc, 2018-11-18:
Do you plan to support [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) in your library?
I tried to support in my fork [mysensors](https://github.com/spectrespc/MySensors)
I get a compilation error:

```
core\core.a(main.cpp.o): In function `premain()':

main.cpp:(.text.startup._Z7premainv+0x0): multiple definition of `premain()'

sketch\RelayActuator.ino.cpp.o:RelayActuator.ino.cpp:(.text.startup._Z7premainv+0x0): first defined here

core\core.a(main.cpp.o): In function `main':

main.cpp:(.text.startup.main+0x0): multiple definition of `main'

sketch\RelayActuator.ino.cpp.o:RelayActuator.ino.cpp:(.text.startup.main+0x0): first defined here
```
Maybe you can point me to my mistake.

I tried to comment all in the file MyMainSTM32.cpp. In the sketch add _begin (); in setup and _process (); in loop. It was possible to compile and load. Noda connected to the gateway, performed the presentation, but did not accept the command, the main loop might have hung up.

sorry google translate...

flatsiedatsie, 2019-01-07:
I'm probably missing what this issue is about, but I've already had MySensors run on an STM32F1 without issue.

tensar, 2019-06-17:
I have the same problems.

platformio.ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = stlink

lib_deps =
  MySensors

To main.cpp I copied RelayActuator.ino from MySensors examples and when I compile it I get 
Linking .pioenvs/genericSTM32F103C8/firmware.elf
.pioenvs/genericSTM32F103C8/src/main.cpp.o: In function `premain()':
main.cpp:(.text.startup._Z7premainv+0x0): multiple definition of `premain()'
.pioenvs/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup._Z7premainv+0x0): first defined here
.pioenvs/genericSTM32F103C8/src/main.cpp.o: In function `main':
main.cpp:(.text.startup.main+0x0): multiple definition of `main'
.pioenvs/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup.main+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/genericSTM32F103C8/firmware.elf] Error 1