mysensors/MySensors #1233: nrf24 lost data

spectrespc, 2018-11-18:
When I try to send a lot of data, some of the data is lost.
I test on 
gateway lolinESP8266 + nrf24l01 nrf24l01 pa lna + 100uF, checked all parameters pa level. Chanel 76/100/10 Lib ver 2.3.1/2.3.0
+
BluePill + nrf24l01 + with/witchout 100uF powered on usb, checked all parameters pa level. Chanel 76/100/10. Lib ver 2.3.1
nrf51822 powered on LIfePO4, checked all parameters pa level. Chanel 76/100/10. Lib ver 2.3.1

Manifestations on presentation, for fix it i add `sleep(200);` . Otherwise, 50% does not reach the destination.

`void presentation()  {
  sendSketchInfo("TEST ", "1.1", true);
  // Déclaration des capteurs attachés au noeud
  present(CHILD_ID_TEMP_1, S_TEMP, "Temp1", true);
  sleep(200);
  present(CHILD_ID_HUM_1, S_HUM, "Hum1", true);
  sleep(200);
  present(CHILD_ID_HUM_STAT_1, S_CUSTOM, "HumStat1", true);
  sleep(200);
  present(CHILD_ID_BARO_1, S_BARO, "Baro1", true);
  sleep(200);
  present(CHILD_ID_FOR_1, S_BARO, "Forecast1", true);
  sleep(300);
  present(CHILD_ID_TEMP_2, S_TEMP, "Temp2", true);
  sleep(200);
  present(CHILD_ID_HUM_2, S_HUM, "Hum2", true);
  sleep(200);
  present(CHILD_ID_HUM_STAT_2, S_CUSTOM, "HumStat2", true);
  sleep(300);
  present(CHILD_ID_LIGHT, S_LIGHT_LEVEL), "Light", true;
  sleep(200);
}`

mfalkvidd, 2018-11-19:
Seems to be the same as 
https://github.com/mysensors/MySensors/issues/892 ?
If so, let's close this issue and collect all discussion in 892.

spectrespc, 2018-11-19:
Ok, close it.