mysensors/MySensors #1242: License

ramack, 2018-12-30:
While browsing through the code I stumbled over the ["GPL v3 or later" statement of the SHA256 implementation for AVR](https://github.com/mysensors/MySensors/blob/03767e34ddf2882952382d672e4bc89711c241aa/hal/crypto/AVR/MyCryptoAVR.cpp#L41)
while the [mysensors code seems to be GPLv2-only](https://github.com/mysensors/MySensors/blob/03767e34ddf2882952382d672e4bc89711c241aa/hal/crypto/AVR/MyCryptoAVR.cpp#L17)

according my understanding GPLv2 and GPLv3+ are incompatible and therefore cannot be linked together.

henrikekblad, 2018-12-30:
Good observation. If you enable sha256 on AVR there will be a license clash. We'll have to discuss how to handle it in the core team (a bit slow over the holidays though).

ramack, 2018-12-30:
no hurry, IANAL and I will not charge you :smile:
Why not changing all to GPLv3+?

While you are at license topics, maybe you can answer my question about the CLA / dual licensing [that I raised in the forum](https://forum.mysensors.org/post/96131)

henrikekblad, 2018-12-30:
Haha.. we're not lawyers either ;) The CLA was put into place for situations just like this. E.g. we wanting move the MySensors library to a GPLv3 license without having to ask all previous contributors.
It also opens up a possibility for a company to embed the MySensors core into some proprietary hardware thingie by using a "commercial" license (hasn't happened yet).

Anyhow. I can conclude that licensing isn't super easy.  Wish I had a little OSS savvy lawyer friend somewhere.

ramack, 2018-12-31:
I'd like to meet that guy too, please 😃
I understand that you want to allow a commercial use, but then maybe it is better to directly select a license without copyleft, or explicitly dual license it from the beginning. Currently for me it has a slight smell, as I as a potential contributor would need to give rights to you without knowing you and without a clear view of what you will or will not do with this right.
The integration of external code is always problematic - with the current CLA and with dual licensing...

ramack, 2019-10-03:
any news about licensing?

ramack, 2022-11-20:
any news? It is not a good state to keep a free software project in a known license clash situation for nearly 4 years!

I strongly recommend to fix that!