mysensors/MySensors #1256: !TSM:FPAR:NO REPLY

cimba007, 2019-01-12:
Hello,

first off I am using an Arduino Pro Mini @ 8MHz as a node.
The SerialGateways connected to iobroke's mysensors plugin via a esp8266 acting as a serial gateway.

This is the log from the gateway (read from bottom to top):
```
https://www.mysensors.org/build/parser?log=close%20details%0Aaccount_circle%0Aadmin%0AAdministrator%0AappsOverview%0AstoreAdapters%0AsubtitlesInstances%0Aview_listObjects%0Aart_trackEnums%0Aview_headlineLog%0Aflash_onEvents%0Aperson_outlineUsers%0AcodeScripts%0AstorageHosts%0Avisibility%0Abuild%0AHOST%20IOBROKER-RASPIioBroker.admin%203.5.10%0Arefresh%20pausedelete%20delete_foreverDOWNLOAD%20LOGLog%20size%200.00%20MB%20%0A%20%20close%09Time%09%09%0AMessage%0Amysensors.0%092019-01-12%2021%3A31%3A39.673%09debug%09Log%20%3A8276138%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%0Amysensors.0%092019-01-12%2021%3A31%3A39.673%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.672%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276138%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.672%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276138%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%0Amysensors.0%092019-01-12%2021%3A31%3A39.580%09debug%09Log%20%3A8276032%20TSF%3AMSG%3AGWL%20OK%0Amysensors.0%092019-01-12%2021%3A31%3A39.580%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.579%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276032%20TSF%3AMSG%3AGWL%20OK%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.579%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276032%20TSF%3AMSG%3AGWL%20OK%0Amysensors.0%092019-01-12%2021%3A31%3A39.578%09debug%09Log%20%3A8276028%20TSF%3ACKU%3AOK%2CFCTRL%0Amysensors.0%092019-01-12%2021%3A31%3A39.577%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.576%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276028%20TSF%3ACKU%3AOK%2CFCTRL%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.576%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276028%20TSF%3ACKU%3AOK%2CFCTRL%0Amysensors.0%092019-01-12%2021%3A31%3A39.575%09debug%09Log%20%3A8276025%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%0Amysensors.0%092019-01-12%2021%3A31%3A39.574%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.573%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276025%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.572%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276025%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%0Amysensors.0%092019-01-12%2021%3A31%3A39.571%09debug%09Log%20%3A8276022%20TSF%3AMSG%3ABC%0Amysensors.0%092019-01-12%2021%3A31%3A39.570%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.570%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276022%20TSF%3AMSG%3ABC%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.569%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276022%20TSF%3AMSG%3ABC%0Amysensors.0%092019-01-12%2021%3A31%3A39.568%09debug%09Log%20%3A8276015%20TSF%3AMSG%3AREAD%2C2-2-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%3A%0Amysensors.0%092019-01-12%2021%3A31%3A39.566%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A39.564%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228276015%20TSF%3AMSG%3AREAD%2C2-2-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%3A%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A39.563%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8276015%20TSF%3AMSG%3AREAD%2C2-2-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%3A%0Amysensors.0%092019-01-12%2021%3A31%3A38.403%09debug%09Log%20%3A8274872%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%0Amysensors.0%092019-01-12%2021%3A31%3A38.402%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A38.401%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228274872%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A38.400%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8274872%20!TSF%3AMSG%3ASEND%2C0-0-2-2%2Cs%3D255%2Cc%3D3%2Ct%3D8%2Cpt%3D1%2Cl%3D1%2Csg%3D0%2Cft%3D0%2Cst%3DNACK%3A0%0Amysensors.0%092019-01-12%2021%3A31%3A37.414%09debug%09Log%20%3A8273863%20TSF%3AMSG%3AGWL%20OK%0Amysensors.0%092019-01-12%2021%3A31%3A37.413%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A37.411%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228273863%20TSF%3AMSG%3AGWL%20OK%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A37.409%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8273863%20TSF%3AMSG%3AGWL%20OK%0Amysensors.0%092019-01-12%2021%3A31%3A37.409%09debug%09Log%20%3A8273859%20TSF%3ACKU%3AOK%2CFCTRL%0Amysensors.0%092019-01-12%2021%3A31%3A37.408%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A37.407%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228273859%20TSF%3ACKU%3AOK%2CFCTRL%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A37.406%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8273859%20TSF%3ACKU%3AOK%2CFCTRL%0Amysensors.0%092019-01-12%2021%3A31%3A37.405%09debug%09Log%20%3A8273856%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%0Amysensors.0%092019-01-12%2021%3A31%3A37.403%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A37.402%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228273856%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A37.400%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8273856%20TSF%3AMSG%3AFPAR%20REQ%2CID%3D2%0Amysensors.0%092019-01-12%2021%3A31%3A37.399%09debug%09Log%20%3A8273853%20TSF%3AMSG%3ABC%0Amysensors.0%092019-01-12%2021%3A31%3A37.398%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A37.396%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228273853%20TSF%3AMSG%3ABC%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A37.395%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8273853%20TSF%3AMSG%3ABC%0Amysensors.0%092019-01-12%2021%3A31%3A37.394%09debug%09Log%20%3A8273846%20TSF%3AMSG%3AREAD%2C2-2-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%3A%0Amysensors.0%092019-01-12%2021%3A31%3A37.392%09debug%09Message%20type%3A%20internal%0Amysensors.0%092019-01-12%2021%3A31%3A37.391%09debug%09Got%3A%20%7B%22id%22%3A%220%22%2C%22childId%22%3A%22255%22%2C%22type%22%3A%22internal%22%2C%22ack%22%3Afalse%2C%22payload%22%3A%228273846%20TSF%3AMSG%3AREAD%2C2-2-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%3A%22%2C%22subType%22%3A%22I_LOG_MESSAGE%22%7D%0Amysensors.0%092019-01-12%2021%3A31%3A37.390%09debug%09Serial%20data%20received%3A%200%3B255%3B3%3B0%3B9%3B8273846%20TSF%3 [... cut ...] 0%20TSM%3AFPAR%0A670093%20TSF%3AMSG%3ASEND%2C2-2-255-255%2Cs%3D255%2Cc%3D3%2Ct%3D7%2Cpt%3D0%2Cl%3D0%2Csg%3D0%2Cft%3D0%2Cst%3DOK%3A%0A672169%20!TSM%3AFPAR%3AFAIL%0A672192%20TSM%3AFAIL%3ACNT%3D7%0A672215%20TSM%3AFAIL%3ADIS%0A672237%20TSF%3ATDI%3ATSL%0A
```

In generel the connection node -> gw seems ok, then the gw want's to ack the fpar request but doesn't get a response.

Could this be related to the fast esp8266 as gw and a "slow" 8mhz node as https://github.com/mysensors/MySensors/issues/578 ?

mfalkvidd, 2019-01-13:
How is the nrf powered? What capacitors are you using?
If you haven't already, see 
https://forum.mysensors.org/topic/666/debug-faq-and-how-ask-for-help/ for the most common problems and how to troubleshoot them efficiently.

cimba007, 2019-01-13:
Thanks for the tipps. It really seems to be power related.
Both nodes are currently powered by a 4,2V lipo battery + 3.3V regulator.

I will add more beefy caps to the regulator.

closed 2019-01-13 by cimba007