mysensors/MySensors #1266: error: macro "min" passed 3 arguments, but takes just 2

andreas-bulling, 2019-02-09:
```
make all && sudo make install
g++ -c -o MyGateway.o MyGateway.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
MyGateway.cpp: In member function ‘void MyGateway::setInclusionMode(boolean)’:
MyGateway.cpp:234:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (newMode != inclusionMode)
   ^~
MyGateway.cpp:237:5: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
     serial(PSTR("0;0;%d;0;%d;%d\n"), C_INTERNAL, I_INCLUSION_MODE, inclusionMode?1:0);
     ^~~~~~
g++ -c -o MySensor.o MySensor.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
g++ -c -o MyMessage.o MyMessage.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
g++ -c -o PiEEPROM.o PiEEPROM.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
g++ -c -o PiGateway.o PiGateway.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
g++ -o PiGateway MyGateway.o MySensor.o MyMessage.o PiEEPROM.o PiGateway.o -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24 -lrf24-bcm
g++ -c -o PiGatewaySerial.o PiGatewaySerial.cpp -Wall -Ofast -lpthread -g -D__Raspberry_Pi -mfloat-abi=hard -D_TTY_NAME=\"/dev/ttyMySensorsGateway\" -D_TTY_GROUPNAME=\"tty\" -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -I. -I/usr/local/include/RF24
In file included from /usr/include/c++/6/bits/char_traits.h:39:0,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/ostream:38,
                 from /usr/include/c++/6/iostream:39,
                 from ./MyMessage.h:37,
                 from ./MySensor.h:18,
                 from ./MyGateway.h:16,
                 from PiGatewaySerial.cpp:36:
/usr/include/c++/6/bits/stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
/usr/include/c++/6/bits/stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
In file included from ./MySensor.h:18:0,
                 from ./MyGateway.h:16,
                 from PiGatewaySerial.cpp:36:
/usr/include/c++/6/bits/stl_algobase.h:195:5: error: expected unqualified-id before ‘const’
     min(const _Tp& __a, const _Tp& __b)
     ^
/usr/include/c++/6/bits/stl_algobase.h:195:5: error: expected ‘)’ before ‘const’
/usr/include/c++/6/bits/stl_algobase.h:195:5: error: expected initializer before ‘const’
/usr/include/c++/6/bits/stl_algobase.h:219:5: error: expected unqualified-id before ‘const’
     max(const _Tp& __a, const _Tp& __b)
     ^
/usr/include/c++/6/bits/stl_algobase.h:219:5: error: expected ‘)’ before ‘const’
/usr/include/c++/6/bits/stl_algobase.h:219:5: error: expected initializer before ‘const’
In file included from /usr/include/c++/6/bits/char_traits.h:39:0,
                 from /usr/include/c++/6/ios:40,
                 from /usr/include/c++/6/ostream:38,
                 from /usr/include/c++/6/iostream:39,
                 from ./MyMessage.h:37,
                 from ./MySensor.h:18,
                 from ./MyGateway.h:16,
                 from PiGatewaySerial.cpp:36:
/usr/include/c++/6/bits/stl_algobase.h:243:5: error: ‘std::min’ declared as an ‘inline’ variable
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^~~
/usr/include/c++/6/bits/stl_algobase.h:246:7: error: expected primary-expression before ‘if’
       if (__comp(__b, __a))
       ^~
/usr/include/c++/6/bits/stl_algobase.h:246:7: error: expected ‘}’ before ‘if’
/usr/include/c++/6/bits/stl_algobase.h:246:7: error: expected ‘;’ before ‘if’
/usr/include/c++/6/bits/stl_algobase.h:248:7: error: expected unqualified-id before ‘return’
       return __a;
       ^~~~~~
/usr/include/c++/6/bits/stl_algobase.h:265:5: error: ‘max’ declared as an ‘inline’ variable
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^~~
/usr/include/c++/6/bits/stl_algobase.h:268:7: error: expected primary-expression before ‘if’
       if (__comp(__a, __b))
       ^~
/usr/include/c++/6/bits/stl_algobase.h:268:7: error: expected ‘}’ before ‘if’
/usr/include/c++/6/bits/stl_algobase.h:268:7: error: expected ‘;’ before ‘if’
/usr/include/c++/6/bits/stl_algobase.h:270:7: error: expected unqualified-id before ‘return’
       return __a;
       ^~~~~~
/usr/include/c++/6/bits/stl_algobase.h:271:5: error: expected declaration before ‘}’ token
     }
     ^
Makefile:61: recipe for target 'PiGatewaySerial.o' failed
make: *** [PiGatewaySerial.o] Error 1
```

mfalkvidd, 2019-02-09:
Thanks for reporting. Are you using the master or development branch?

mfalkvidd, 2019-02-09:
@andreas-bulling MyGateway.cpp does not exist in this repository. What version are you using?

andreas-bulling, 2019-02-10:
https://forum.mysensors.org/topic/8932/raspberry-pi3-gateway/5

andreas-bulling, 2019-02-10:
latest version from repository I believe...

mfalkvidd, 2019-02-10:
@andreas-bulling did you read the first reply in the thread you linked?

andreas-bulling, 2019-02-10:
yes, when I found the link 6 hours ago ;)

closed 2019-02-10 by mfalkvidd