mysensors/MySensors #1276: Old versions on PlatformIO

pvojnisek, 2019-04-15:
I am migrating my projects to PlatformIO. As an old MySensors user I still have nodes (and networks) running on MySensors 1.5.x versions. Unfortunately earlier versions (older than 2.1.0) of MySensors library are not published to PlatformIO Library Manager. Would you please check if you can publish them? Thank you!

mfalkvidd, 2019-04-16:
I think publishing MySensors to the Arduino Library Manager required moving/changing 1000+ files (see https://github.com/mysensors/MySensors/commit/ba6bb915c2c141ee3b3bb80b7fb872e5f1063ac9#diff-5a96806328fdf598e651d77cf92879fe ). I think Platformio uses the same mechanism as the Arduino Library Manager. So the effort is likely big, but if someone wants to have a go at it we can probably handle it if it is done in a separate branch to avoid affecting the working 2.x code.

mfalkvidd, 2019-04-16:
It seems a platformio file was present for 1.5.3 and 1.5.4 (see https://github.com/mysensors/MySensors/commit/1f27c3dd36ea687092de115fea1b9c8dcd9a6928#diff-d5246ba0cf6ddec25b7abbfb1f8dbcd1 )
Doesn't that mean that 1.5.3 and 1.5.4 should already be available in platformio?

pvojnisek, 2019-04-29:
It's strange. You can check Platformio library here: https://platformio.org/lib/show/548/MySensors
`**Versions**
2.3.1 released 4 months ago
2.3.0 released 10 months ago
2.2.0 released a year ago
2.1.1 released over a year ago
2.1.0 released over a year ago`
If it's a huge work with lots of file movement do not spend your time on it. But if there is a configuration issue, please consider adding the old versions.