mysensors/MySensors #1277: Add push button type

flatsiedatsie, 2019-04-19:
I am really missing a built-in push button type.

I was again triggered to create this request while working on a mysensors adapter for the Mozilla Gateway. It has the 'push-button' capability. But I cannot map that to anything in MySensors.

Earlier I noticed that in my sketches I wish I could more clearly have a pushbutton child. For example, in Domoticz a user now has to manually switch a button onto a pushbutton type via the interface. It would be great if this intention could already be built in.

mfalkvidd, 2019-04-19:
@flatsiedatsie do I understand correctly that you mean https://iot.mozilla.org/schemas/#PushButton ?
Would it make sense to map that to MySensors as S_BINARY with V_TRIPPED?

How are the other Mozilla types mapped to MySensors? (I guess it would make sense to look at S_BINARY with V_STATUS as a comparison.)

flatsiedatsie, 2019-04-22:
> Would it make sense to map that to MySensors as S_BINARY with V_TRIPPED?

I don't know. Would that work? I'd just like to signal to the controller interface that the button should immediately return to the unpressed state after it is pressed. Right now my nodes 'hack' it by sending a "go back to unpressed state" update after they receive the signal. But in some scenario's that creates unforeseen complexity. And if the return signal is not received, then the controller UX is left in a weird/impossible state.

I'll share a first version of the adapter soon (possibly today), there you can see the initial mapping (still very incomplete, but it works for a few device types now)

flatsiedatsie, 2019-04-28:
Getting back to this: do you think most MySensors users would find the behaviour or an S_BINARY with V_TRIPPED being turned into a pushbutton predictable?

If so, then I can make this the behaviour of the WebThings adapter.

henrikekblad, 2019-04-28:
I would suggest using S_SCENE_CONTROLLER which usually is a panel with multiple push buttons.
The corresponding V_SCENE_ON is triggered when button is pressed. Some scene controllers use V_SCENE_OFF when doing a long press.

flatsiedatsie, 2019-04-28:
Hmm, I was just thinking that  V_TRIPPED feels more like sensor data than an actuator.

The S_SCENE_CONTROLLER option could work, but feels again a bit like a hack.

Which would again make the point that a dedicated push button type would be really useful?

Just thinking out loud: how about this situation:

S_BINARY <- handles toggle buttons
- V_STATUS. Things like 'long push off' or 'long push on' could just be sent as the number 3 and 4? Double-press off and on could be 5 and 6? Etc.
- V_READ_ONLY <- This would communicate if the button can be toggled from the controller side, or if it's just an indication of the button status on the MySensors device. Can be 0 or 1. If set to 0 it effectively communicates "I am a binary sensor".
 - V_UNIT<- As with other S_TYPES in MySensors, his would give meta-information about the child. In this case it would indicate if the child is of a certain type, which can be used to optimize the presentation in the UX. For example: "lock", "door", "alarm".

S_NOTIFY / S_EVENT / S_UNARY / S_MONO / S_PUSH / S_SINGLE <- This handles signals that are not binary, but indicate an event has occurred. Such as push buttons.
- V_STATUS <- The message that the event just occurred. Can be 0 or 1, whichever the user wants. Things like 'long push' could just be sent as the number 3 and 4? Double-press could be 5 and 6? etc.
- V_READ_ONLY <- Same as above.
- V_UNIT <- Same as above. Could be "push", "alert", whatever is agreed upon.

You could create pretty much al singular and binary actuators and sensors with this.

mfalkvidd, 2019-04-28:
Does anyone know how popular controllers like OpenHab and HomeAssistant handle this type of button? Or don't they support this use case!

MartinHjelmare, 2019-04-28:
In Home Assistant, for now, we treat sensors that report V_TRIPPED as binary sensors with state. We don't  specifically support push buttons although S_SCENE_CONTROLLER + V_SCENE_ON can be used for that.

I'm planning to fire Home Assistant events for binary sensors, on child update, besides the normal state changes, to allow users to write automations that listen for events and not only state changes. In Home Assistant stateless devices are usually represented as events that fire.

If mysensors would be more specific in reporting if a button has state or is stateless, I think that would allow for a better user experience, where controllers can present the devices and their behavior more close to what they actually are.

MartinHjelmare, 2019-07-18:
It would probably involve adding a new S type and one to many V types for the different push events. Probably at least two different push events, click and hold (long press) should be represented.

I haven't given this much thought yet.