mysensors/MySensors #1283: Feature request: MQTT over SSL/TLS

turboproc, 2019-04-23:
As far as I understand the current MQTT support does not take into consideration security by means of TLS/SSL support (between the gateway and MQTT broker).
Question is whether this has been planned already or is on someone's list.

Looking at the current code where a standard MQTT library is used not supporting TLS, probably a change of library would be required that supports this out of the box, like the PaHo MQTT lib.

Any thoughts on this?

rdrgzlng, 2019-12-27:
I have a modification of the _"MyGatewayTransportMQTTClient.cpp"_ file to support the SSL / TLS connection.

At this time, it is necessary to define three constants, in a _"certificates.h"_ file, for example:

```
const char mqtt_ca_cert[] PROGMEM = R"EOF(
----- BEGIN THE CERTIFICATE -----
XXX ... XXX
----- FINISH CERTIFICATE -----
)EOF";

const char mqtt_client_cert[] PROGMEM = R"EOF(
----- BEGIN THE CERTIFICATE -----
XXX ... XXX
----- FINISH CERTIFICATE -----
)EOF";

const char mqtt_client_key[] PROGMEM = R"EOF(
----- START THE RSA PRIVATE KEY -----
XXX ... XXX
----- FINISH THE RSA PRIVATE KEY -----
)EOF";
```

And put three new #defines in the sketch:

```
#define MY_MQTT_CA_CERT mqtt_ca_cert
#define MY_MQTT_CLIENT_CERT mqtt_client_cert
#define MY_MQTT_CLIENT_KEY mqtt_client_key
```

If you want, I can do a pull request with these changes.

In the future, SPIFS can be used to store certificates if you think it's interesting.

rdrgzlng, 2019-12-27:
With these changes, it is possible to connect a ESP8266 MQTT Gateway with AWS IoT using certificates.

closed 2020-03-08 by tekka007