mysensors/MySensors #1284: V_Unit_Prefix for S_weight?

flatsiedatsie, 2019-04-24:
Shouldn't users be able to set what weight scale they are measuring in? 

For example: Kilograms, grams, or even non-metric things?

Or is that perhaps what that 'V_Impedance' is for?

mfalkvidd, 2019-04-25:
For S_TEMP / V_TEMP, the unit is decided by the controller's isMetric setting.
Reference https://github.com/mysensors/MySensors/blob/a1282f358aadfe3dcd5b486778274a8ddc869ac5/core/MySensorsCore.h#L89

Would it make sense to use something similar for S_WEIGHT? Or should units be handled in some other way? The current implementation for temp does't support Kelvin for example, so maybe the current implementation is i sufficient?

mfalkvidd, 2019-04-25:
Are there any specifications we could adapt, instead of inventing our own?
OMA LightweightM2M has [this specification for a "load" sensor](http://www.openmobilealliance.org/tech/profiles/lwm2m/3322.xml). They use a free-form string for specifying the unit. That approach is flexible, but maybe too flexible? Could we find other useful specifications?

mfalkvidd, 2019-04-25:
And no, impedance is not related. This is what impedance is used for:

> In addition to measuring weight, the bathroom scale calculates body composition by sending a mild electrical impulse through the body. The measured electrical resistance is used – together with information on age, gender and height – to calculate body fat, body water, and more.

flatsiedatsie, 2019-04-25:
Ah, so that's what its for! Interesting.

I thought it would be useful to just add `V_UNIT_PREFIX`? That way the device can specify in what scale it's measuring (grams/kilograms).

mfalkvidd, 2019-04-25:
Sorry, I thought you proposed a new type, but I just noticed that V_UNIT_PREFIX is already used for distance, air quality and dust level.

Yes, using V_UNIT_PREFIX for any value that can have different units would be quite useful. In addition to the existing uses and weight (as per your suggestion) I think it would make sense to also use V_UNIT_PREFIX for things like voltage, current, atmospheric pressure, power usage (although V_KWH unfortunately already implies a unit), flow and volume.

flatsiedatsie, 2019-04-28:
Exactly!

See also [this issue](https://github.com/mysensors/MySensors/issues/1277) for some musing on making some universal V_TYPES to help presentation in the UX

mfalkvidd, 2019-07-18:
See also https://github.com/mysensors/MySensors/issues/1312 for yet another discussion on the same topic.

flatsiedatsie, 2019-07-20:
Yes I thought I'd create an official proposal for it.