mysensors/MySensors #1299: sendSketchInfo => I_SKETCH_NAME missing with #define MY_DEBUG

frenchclem, 2019-06-13:
Hi,
I have a strange behavior with sendSketchInfo  fonction. 
I reproduced the problem with the motion sensor exemple. 

If I enable  #define MY_DEBUG => the presentation is complet (I_SKETCH_NAME  + I_SKETCH_VERSION)
If I disable  #define MY_DEBUG => the presentation is UNcomplet (I_SKETCH_NAME IS MISSING + I_SKETCH_VERSION is Visible) 

I couldn't reproduce the bug with the passive node exemple. 

With the sendSketchInfo fonction below I don't understand how this could be possible ?!?

bool sendSketchInfo(const char *name, const char *version, const bool echo)
{
	bool result = true;
	if (name) {
		result &= _sendRoute(build(_msgTmp, GATEWAY_ADDRESS, NODE_SENSOR_ID, C_INTERNAL, I_SKETCH_NAME,
		                           echo).set(name));
	}
	if (version) {
		result &= _sendRoute(build(_msgTmp, GATEWAY_ADDRESS, NODE_SENSOR_ID, C_INTERNAL, I_SKETCH_VERSION,
		                           echo).set(version));
	}
	return result;
}

I'm using MYSController for debug. My gateway and node a programmed with MySensors 2.3.1 version.

frenchclem, 2019-06-13:
Little update here with my issue 

* I'm using NRF52
* I ran a second test suspecting #define MY_DEBUG to slow down the MCU - I added a sendSketchInfo("Motion Sensor", "1.0"); - see below my presentation() 

On the first sendSketchInfo("Motion Sensor", "1.0");  I receive only the I_SKETCH_VERSION
On the second one I get everything as expected.

I looks like the first msg with I_SKETCH_NAME is never sent ?!?

See below the logs
13/06/2019 09:22:47	RX	75;255;3;0;6;0
13/06/2019 09:22:47	TX	75;255;3;0;6;M
13/06/2019 09:22:47	RX	0;255;3;0;9;54274304 !TSF:MSG:SEND,0-0-75-75,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=NACK:M
13/06/2019 09:22:47	RX	0;255;3;0;9;54274312 TSF:MSG:READ,75-75-0,s=255,c=3,t=12,pt=0,l=3,sg=0:1.0
13/06/2019 09:22:47	RX	75;255;3;0;12;1.0
13/06/2019 09:22:47	RX	0;255;3;0;9;54274319 TSF:MSG:READ,75-75-0,s=255,c=3,t=11,pt=0,l=13,sg=0:Motion Sensor
13/06/2019 09:22:47	RX	75;255;3;0;11;Motion Sensor
13/06/2019 09:22:47	RX	0;255;3;0;9;54274328 TSF:MSG:READ,75-75-0,s=255,c=3,t=12,pt=0,l=3,sg=0:1.0
13/06/2019 09:22:47	RX	75;255;3;0;12;1.0
13/06/2019 09:22:47	RX	0;255;3;0;9;54274337 TSF:MSG:READ,75-75-0,s=1,c=0,t=1,pt=0,l=0,sg=0:
13/06/2019 09:22:47	RX	75;1;0;0;1;

void presentation()
{
  // Send the sketch version information to the gateway and Controller
  sendSketchInfo("Motion Sensor", "1.0");
  sendSketchInfo("Motion Sensor", "1.0");
  // Register all sensors to gw (they will be created as child devices)
  present(CHILD_ID, S_MOTION);
}

mfalkvidd, 2019-06-14:
The node log will give more details.
Most likely the first sketch name message is lost (=sent by the node but not received by the gateway). As indicated by st=NACK:M there are communication problems.