mysensors/MySensors #1300: MinorChanges: SoftwareSerial for RS485

ploeffler, 2019-06-14:
Todo:
+) include SoftwareSerial to drivers
+) include espsoftwareserial to drivers

Changes in File MySensors.h

replace line 372 with
----
#if defined(MY_RS485_SWSERIAL)
#if defined(ESP8266) || defined(ESP32)
#include "drivers/ESPSoftwareSerial/SoftwareSerial.cpp"
#elif 
#include "drivers/SoftwareSerial/SoftwareSerial.cpp"
#endif 
#elif 
#include "drivers/AltSoftSerial/AltSoftSerial.cpp"
#endif 
-------

File hal/transport/RS485/MyTransportRS485.cpp
replace line 90 with:
-----
#if defined(MY_RS485_SWSERIAL)
SoftwareSerial& _dev  = MY_RS485_SWSERIAL;
#elif 
#include "drivers/AltSoftSerial/AltSoftSerial.cpp"
#endif 
------

thank you!

mfalkvidd, 2019-06-15:
Thanks for your suggestion.

AVR (and maybe others) already use soft serial for rs485 by default, so there is no need to introduce a new define.

> include SoftwareSerial to drivers

Could you describe which software serial you mean?

> include espsoftwareserial to drivers

Do you mean https://github.com/plerup/espsoftwareserial ? (There are multiple libraries matching that name and it would be good if we knew which one to use.)

For reference, AltSoftSerial (which is currently used if a RS485 node or gateway is compiled for ESP8266/ESP32 with MySensors) provides this output during compilation:
```
In file included from ...\Arduino\libraries\MySensors/drivers/AltSoftSerial/AltSoftSerial.cpp:35:0,

                 from ...\Arduino\libraries\MySensors/MySensors.h:372,

                 from ...\Arduino\libraries\MySensors\examples\GatewaySerialRS485\GatewaySerialRS485.ino:89:

...\Arduino\libraries\MySensors/drivers/AltSoftSerial/config/AltSoftSerial_Boards.h:146:2: error: #error "Please define your board timer and pins"

 #error "Please define your board timer and pins"

  ^

Using library MySensors at version 2.3.2-beta in folder: ...\Arduino\libraries\MySensors
```

ploeffler, 2019-06-15:
sorry, no experiences with Altsoftserial, except the same you had (and had no motivation to read that part of the code too)

SoftwareSerial: i have read the discussions (some months ago)  with the changes in the source too, its really not important WHICH of them to be used. just keep in mind that ESP's need another one then the AVR's