mysensors/MySensors #1311: Diagnostic logs using PRIu8 family of macros broken

pbolduc, 2019-07-17:
There are a number of macros defined in C/C++ that provide portal format strings for things like **printf**.  However, there is an issue on microcontrollers and  gcc-arm-embedded toolchain.

This is related to this pull request https://github.com/mysensors/MySensors/pull/943 which fixed it for NRF5.  However, the problem is related to any platform that uses the gcc-arm-embedded toolchain, like Arduino IDE or platform IO and a number of small microcontrollers.

I had raised an issue with the GCC here: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1836698 to try to determine what was going on.  It seems these embedded processors will use newlib-nano instead the full newlib for formatting of strings. See https://community.arm.com/developer/ip-products/system/b/embedded-blog/posts/shrink-your-mcu-code-size-with-gcc-arm-embedded-4-7

_Libraries also need optimizing, because the libraries included in GCC ARM Embedded were not actually designed for MCU programming. Newlib, the C library in the toolchain, implements printf functions that are so complicated they require about 37K bytes of FLASH and 5K bytes of RAM to run a simple hello-world program. That's far too large for MCU programming where you might need printf functionality for debugging and logging purposes. The good news is that there is plenty of unnecessary "fat" in libraries that can be cut._

_Newlib-nano cuts some features that were added after C89, which are believed to be rarely used in MCU programming. By limiting the format converter to the C89 standard, format string processing code in printf is greatly reduced._

_To summarize, the newlib-nano can cut the size of hello-world programs by around 80%. In extreme cases for C++ programs, the size reduction could exceed 90%._

I think the mysensors could be changed to handle this globally.  Perhaps adding an addition MY_ABC kind of pre-compiler setting that would enable this.  Could also check various ARDUINO_SAMD_ZERO ARDUINO_ARCH_SAMD 

Ideally, I would prefer if the framework could detect this automatically and apply the redefines but I do not know enough about the build process.

Thoughts?

pbolduc, 2019-07-24:
I just noticed in the development branch has this [commit](https://github.com/mysensors/MySensors/pull/1302/commits/fd99de020a8a4b0df3d58e74f413974149bb5f93) that was part of pull request #1302 which resolves this issue.

mfalkvidd, 2019-07-25:
Thanks @pbolduc 

I'll close this issue then. Just reopen if there is something missing.

closed 2019-07-25 by mfalkvidd