mysensors/MySensors #1312: Proposal: UX extensions

flatsiedatsie, 2019-07-18:
While creating a MySensors add-on for the Mozilla WebThings Gateway I gained a lot of understanding of what MySensors can do. But I also got a better understanding of what could still be improved.

Specifically, I feel that MySensors devices could communicate better about how their capabilities might be displayed in the user interface of smart home controllers.

To make this very concrete, I want to propose this:

**V_READONLY**
Make it an official specification that _all_ S_types can have the V_Type `V_READONLY`. If the node sends a 1 for this V_Type, then the UI will know that the value cannot be modified by the user and returned to the node. This is useful for sensors, but it also allows for differentiation on when an S_INFO should be shown as an input field, or if it should be shown as an simple text output only. It's also useful for S_BINARY children, where you could then tell the UI to show a toggle or a simple state output.

**V_UNIT**
Make it an official specification that _all_ S_types can have the V_Type `V_UNIT`. This will allow the UX to better explain what the variable is. For a distance sensor, it could communicate 'mm' or 'cm'. For binary toggles it could communicate what the UI should show. For example: 'on/off' or 'open/closed'.

mfalkvidd, 2019-07-18:
How would V_UNIT differ from V_UNIT_PREFIX ?

flatsiedatsie, 2019-07-18:
Haha, I meant that one! I'll update it in the first post.

mfalkvidd, 2019-09-09:
Related question in the forum where V_READ_ONLY could be applicable: https://forum.mysensors.org/topic/10631/what-is-proper-presentation-type-for-binary-push-button-s_binary-and-how-make-controller-distinguish-between-binary-button-ro-and-binary-switch-rw/