mysensors/MySensors #1314: Error while compiling on Raspberry Pi 4

flatsiedatsie, 2019-07-20:
This is the complete log from a Raspberry Pi 4 running the latest version of Buster.

```
pi@gateway:~/MySensors$ ./configure
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=unknown, Type=unknown, CPU=armv7l.
[SECTION] Checking GPIO Sysfs.
  [OK] /sys/class/gpio/export found.
[SECTION] Detecting SPI driver.
  [OK] SPI driver detected:SPIDEV.
[SECTION] Gateway configuration.
  [OK] Type: ethernet.
  [OK] Transport: rf24.
  [OK] Signing: Disabled.
  [OK] Encryption: Disabled.
  [OK] CPPFLAGS:  -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV 
  [OK] CXXFLAGS:  -std=c++11
[SECTION] Detecting init system.
  [OK] Init system detected: systemd.
[SECTION] Saving configuration.
  [OK] Saved.
[SECTION] Cleaning previous builds.
  [OK] Finished.
pi@gateway:~/MySensors$ make
gcc -MT build/hal/architecture/Linux/drivers/core/log.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/log.c -o build/hal/architecture/Linux/drivers/core/log.o
gcc -MT build/hal/architecture/Linux/drivers/core/config.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/config.c -o build/hal/architecture/Linux/drivers/core/config.o
g++ -MT build/hal/architecture/Linux/drivers/core/StdInOutStream.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/StdInOutStream.cpp -o build/hal/architecture/Linux/drivers/core/StdInOutStream.o
g++ -MT build/hal/architecture/Linux/drivers/core/IPAddress.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/IPAddress.cpp -o build/hal/architecture/Linux/drivers/core/IPAddress.o
g++ -MT build/hal/architecture/Linux/drivers/core/noniso.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/noniso.cpp -o build/hal/architecture/Linux/drivers/core/noniso.o
g++ -MT build/hal/architecture/Linux/drivers/core/GPIO.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/GPIO.cpp -o build/hal/architecture/Linux/drivers/core/GPIO.o
hal/architecture/Linux/drivers/core/GPIO.cpp: In constructor ‘GPIOClass::GPIOClass()’:
hal/architecture/Linux/drivers/core/GPIO.cpp:53:18: warning: ‘%s’ directive writing up to 255 bytes into a region of size 48 [-Wformat-overflow=]
    sprintf(file, "/sys/class/gpio/%s/base", de->d_name);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:53:11: note: ‘sprintf’ output between 22 and 277 bytes into a destination of size 64
    sprintf(file, "/sys/class/gpio/%s/base", de->d_name);
    ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:62:18: warning: ‘%s’ directive writing up to 255 bytes into a region of size 48 [-Wformat-overflow=]
    sprintf(file, "/sys/class/gpio/%s/ngpio", de->d_name);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:62:11: note: ‘sprintf’ output between 23 and 278 bytes into a destination of size 64
    sprintf(file, "/sys/class/gpio/%s/ngpio", de->d_name);
    ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++ -MT build/hal/architecture/Linux/drivers/core/SPIDEV.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/SPIDEV.cpp -o build/hal/architecture/Linux/drivers/core/SPIDEV.o
g++ -MT build/hal/architecture/Linux/drivers/core/Print.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/Print.cpp -o build/hal/architecture/Linux/drivers/core/Print.o
g++ -MT build/hal/architecture/Linux/drivers/core/EthernetClient.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/EthernetClient.cpp -o build/hal/architecture/Linux/drivers/core/EthernetClient.o
g++ -MT build/hal/architecture/Linux/drivers/core/compatibility.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/compatibility.cpp -o build/hal/architecture/Linux/drivers/core/compatibility.o
g++ -MT build/hal/architecture/Linux/drivers/core/SerialPort.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/SerialPort.cpp -o build/hal/architecture/Linux/drivers/core/SerialPort.o
g++ -MT build/hal/architecture/Linux/drivers/core/Stream.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/Stream.cpp -o build/hal/architecture/Linux/drivers/core/Stream.o
g++ -MT build/hal/architecture/Linux/drivers/core/interrupt.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/interrupt.cpp -o build/hal/architecture/Linux/drivers/core/interrupt.o
g++ -MT build/hal/architecture/Linux/drivers/core/SoftEeprom.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/SoftEeprom.cpp -o build/hal/architecture/Linux/drivers/core/SoftEeprom.o
g++ -MT build/hal/architecture/Linux/drivers/core/EthernetServer.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c hal/architecture/Linux/drivers/core/EthernetServer.cpp -o build/hal/architecture/Linux/drivers/core/EthernetServer.o
g++ -MT build/examples_linux/mysgw.o -MMD -MP -DMY_RADIO_RF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_SPIDEV  -Ofast -g -Wall -Wextra -std=c++11 -I. -I./core -I./hal/architecture/Linux/drivers/core -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from ./MySensors.h:431,
                 from examples_linux/mysgw.cpp:82:
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘bool transportSendWrite(uint8_t, MyMessage&)’ at ./core/MyTransport.cpp:989:2,
    inlined from ‘bool transportRouteMessage(MyMessage&)’ at ./core/MyTransport.cpp:561:40,
    inlined from ‘void stUplinkTransition()’ at ./core/MyTransport.cpp:252:29:
./core/MyMessage.cpp:281:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncatio]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘char* protocolMyMessage2Serial(MyMessage&)’ at ./core/MyProtocol.cpp:88:8:
./core/MyMessage.cpp:281:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncatio]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘char* protocolMyMessage2Serial(MyMessage&)’ at ./core/MyProtocol.cpp:88:8,
[... cut ...] /EthernetClient.o build/hal/architecture/Linux/drivers/core/compatibility.o build/hal/architecture/Linux/drivers/core/SerialPort.o build/hal/architecture/Linux/drivers/core/Stream.o build/hal/architecture/Linux/drivers/core/interrupt.o build/hal/architecture/Linux/drivers/core/SoftEeprom.o build/hal/architecture/Linux/drivers/core/EthernetServer.o build/examples_linux/mysgw.o 
```

The stable code gives less errors. See: https://forum.mysensors.org/topic/4803/building-a-raspberry-pi-gateway/1019

mfalkvidd, 2019-08-24:
Not sure if this is related, or needed, but the upstream nrf24 library has some updates for handling rpi4 pullups. https://github.com/nRF24/RF24/compare/v1.3.3...master

jl277013, 2019-08-24:
I have the same problem.  I have tried both the branch master and development.  Will have to keep the RBP3 up until this is resolved.

flatsiedatsie, 2019-08-24:
@jl277013 Or use a serial gateway.

jl277013, 2019-08-24:
@flatsiedatsie , I already have the RBP hat for the nRF24.   It should just be a software fix and no need to build a new gateway.

flatsiedatsie, 2019-08-29:
Of course. I meant as a temporary solution.