mysensors/MySensors #1320: Wifi disconnect on ESP gateways

rikki78, 2019-07-23:
Hi
I've been struggling with a ESP32/ESP8266 issue for a while, and that is that it in some WiFi-reconnect cases, doesn't reconnect automatically. I've been able to reproduce this by applying new settings to my router, which makes ESP based gateways disconnect.
The point is that they don't reconnect automatically, and the host application is expected to check on this itself. This should be implemented in MySensors also, to make the system more stable.

mfalkvidd, 2019-07-23:
My esp8266 gw (rfm69, tcp server mode) reconnects if I reconfigure my router (causing reset of the router's wifi). The gw  also reconnects when I reboot my router. So there must be some difference between your setup and my setup.

What router do you have?
What gateway settings do you use?
What does the gateway's debug log saybefore, during and after the disconnect?

rikki78, 2019-07-23:
I have a Asus rt-ac66u, but I have also used my cable modem router (a Compal CH7465LG-ZG). 
I'm using a MQTT gateway. The debug log shows something like 
GWT:TPC:CONNECTING... 
but never comes back to connect.
When I add some lines of code to reconnect (using WifiManager library) it reconnects within a few moments and everything works again.
If you want I can replicate the issue later this week to provide full logs

mfalkvidd, 2019-07-24:
@rikki78 could be same problem as https://github.com/mysensors/MySensors/issues/788
Which version of MySensors are you using? If it is the same problem as 788, a fix is available in the development version of MySensors. You could try installing it and see if the problem goes away.

rikki78, 2019-07-31:
Sorry not coming back to this. I tried to replicate this but I have some issues with connecting to my router and can't replicate it right now.
I'm running the development version of MySensors so 2.4.2 beta. 
The problem is different than the #788 you indicate, as it keeps running but doesn't reconnect. I think it is more a ESP issue than a MySensors issue, but the framework could prevent it from happening.

rikki78, 2019-09-22:
Well, after a lot of debugging and trying to find what's going on, I think this is related to the ESP32 specifically. When I use the latest MySensors ESP32 MQTT gateway example, it won't connect to my network. I've tried to have it connect to another router also, but that gave the same result. 
Using other connect software (like WifiManager) it does work. So I've found out that when I add a `delay(250); `
on line 261 of MyGatewayTransportMQTTClient.cpp, it does connect to my network. And, it also reconnects to the network if I change a setting on the network.
So I will try to submit a PR for this.

rikki78, 2019-09-23:
By switching to version 1.10 of espressif framework the issue is resolved

closed 2019-09-23 by rikki78