mysensors/MySensors #1334: Fix ESP8266/ESP32 "ISR not in IRAM" runtime exception

Yveaux, 2019-08-10:
Refer to: https://forum.mysensors.org/topic/10581/isr-not-in-iram-error

ESP interrupt handlers shall be prefixed with "ICACHE_RAM_ATTR"

mfalkvidd, 2019-08-10:
Since this attribute was not required in earlier board versions, maybe we need to check if the adding the attribute will break things for older board versions.

mfalkvidd, 2019-08-11:
The incompatible change in the esp8266 board definition was introduced in v2.5.1, so a workaround is to use board version 2.5.0 or earlier.

Reference: https://github.com/esp8266/Arduino/pull/6141

mfalkvidd, 2019-08-11:
Also see https://groups.google.com/a/arduino.cc/forum/m/#!topic/developers/56VPC9t2MPc which discusses how to generalize the ICACHE_RAM_ATTR requirement. It mentions thatt all functions called from within an ISR also needs the attribute. If that is true, a lot of funcrions will be affected.

closed 2019-09-24 by mfalkvidd

sp9auv, 2023-05-25:
Problem with ISR not IRAM in a WEMOS or ARDUION is that  memory for  that handle ISR interrupt must be reserved in a section DECLARATION for values  in this way for example 
**void ICACHE_RAM_ATTR ServiceLimitSwitchUpperShield_12_D6();**   
 // There are solved for  ATTACHINTERRUPT - you must **declare this before function SETUP (very important )** 
In a section SETUP you can define 
**attachInterrupt(digitalPinToInterrupt(12), ServiceLimitSwitchUpperShield_12_D6 , FALLING );**
      
The method ServiceLimitSwitchUpperShield_12_D6   define for example
**[[ichache_IRAM_ATTR]] 
void ServiceLimitSwitchUpperShield_12_D6()** 

Take attention what memory is need for your souce code . This value give compilator.With abouve 32 KB may also occur problem . Choice MMU from Tools 
as 32 KB cache + 32 KB IRAM (balanced ). You can try other option 
de sp9auv

PR #1348 (2019-09-24): ESP32/ESP8266: place ISR in IRAM
- Fixes #1334