mysensors/MySensors #1337: Add time measurement

Strontvlieg, 2019-08-13:
Hi There,

Is it possible to add the following options:

Type: S_TIME 
Variables: V_TIMEMS, V_TIMESEC, V_TIMEMIN and V_TIMEHOUR or something that looks like it

This gives the possibility to display a measured time.
Thank you for this beautiful piece of software.

Greetz

flatsiedatsie, 2019-09-27:
I was about to post the _exact_ same suggestion!

This would be very useful for making (alarm)clocks.

matkor, 2020-02-22:
I volunteer to make pull request with examples (and later test it in Home Assistant) once I get green light from project managers.
I propose pick one (or two) V_*s to reduce complication level.
For my needs V_TIMEMS  representing  signed number of miliseconds using 32bits (which gives more -24 to 24 days values) representing periods/interval of time/delta time/ relative events /elapsed time if fine and enough.
Positive value for future events/periods/intervals, negative value for past events, elapsed time (?) 

I need time based autonomous switches/relays for garden watering, irrigation sprinklers, move detector lights, so I can set them:  turn on/off  for different configurable periods and they will turn off autonomously without controller.

matkor, 2020-02-22:
RO/RW problem will also affect S_TIME as mentioned in https://github.com/mysensors/MySensors/issues/1312