mysensors/MySensors #1342: Pi Gateway crashes upon connection (latest dev branch)

Swiftnesses, 2019-09-05:
Hi,

After installing the latest dev branch on my Pi, I now get the following error when Home Assistant tries to connect (the mys process is then closed).

```
Sep 05 09:06:34 DEBUG New connection from 192.168.1.16
Sep 05 09:06:34 DEBUG GWT:TSA:C=2,CONNECTED
Sep 05 09:06:34 DEBUG GWT:RFC:C=2,MSG=255;255;3;0;20;
Segmentation fault
```

Swiftnesses, 2019-09-05:
@tekka007 I've scoured the recent commits but can't seem to find the issue. I last installed the dev branch 6 months or so back, it worked then.

Interestingly, connecting via MYS Controller doesn't cause a crash and uses:

`DEBUG GWT:RFC:C=2,MSG=255;255;3;0;20;0`

Note the extra '0' on the end, related?

Swiftnesses, 2019-09-05:
Output using strace:

```
gettimeofday({tv_sec=1567677053, tv_usec=800246}, NULL) = 0
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=127, ...}) = 0
write(2, "Sep 05 09:50:53 \33[36mDEBUG\33[0m ", 31Sep 05 09:50:53 DEBUG ) = 31
write(2, "GWT:RFC:C=1,MSG=255;255;3;0;20;\n", 32GWT:RFC:C=1,MSG=255;255;3;0;20;
) = 32
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

recipedude, 2019-09-05:
Have the same issue, however it seems intermittent. 

Here's console output from a manual run showing a controller connect that doesn't fail, and later a segmentation fault. 

```pi@raspi:/usr/local/bin $ sudo ./mysgw
Sep 05 08:56:51 INFO  Starting gateway...
Sep 05 08:56:51 INFO  Protocol version - 2.3.2-beta
Sep 05 08:56:51 DEBUG MCO:BGN:INIT GW,CP=RNNGL---,FQ=NA,REL=7,VER=2.3.2-beta
Sep 05 08:56:51 DEBUG TSF:LRT:OK
Sep 05 08:56:51 DEBUG TSM:INIT
Sep 05 08:56:51 DEBUG TSF:WUR:MS=0
Sep 05 08:56:51 DEBUG TSM:INIT:TSP OK
Sep 05 08:56:51 DEBUG TSM:INIT:GW MODE
Sep 05 08:56:51 DEBUG TSM:READY:ID=0,PAR=0,DIS=0
Sep 05 08:56:51 DEBUG MCO:REG:NOT NEEDED
Sep 05 08:56:51 DEBUG Listening for connections on �U�~:5003
Sep 05 08:56:51 DEBUG MCO:BGN:STP
Sep 05 08:56:51 DEBUG MCO:BGN:INIT OK,TSP=1
Sep 05 08:56:51 DEBUG TSM:READY:NWD REQ
Sep 05 08:56:51 DEBUG ?TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
Sep 05 08:56:52 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:29
Sep 05 08:56:58 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:57:09 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:91
Sep 05 08:57:15 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:57:27 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:63
Sep 05 08:57:32 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:57:46 DEBUG New connection from 10.0.0.157
Sep 05 08:57:46 DEBUG GWT:TSA:C=0,CONNECTED
Sep 05 08:58:01 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:82
Sep 05 08:58:07 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:58:18 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:45
Sep 05 08:58:24 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:58:58 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:5
Sep 05 08:59:04 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:59:27 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:54
Sep 05 08:59:33 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:98
Sep 05 08:59:39 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:26
Sep 05 08:59:44 DEBUG TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=0:102
Sep 05 08:59:46 DEBUG GWT:RFC:C=0,MSG=0;255;3;0;2;
Segmentation fault```

recipedude, 2019-09-05:
Repeated running strace.  Failed as soon as the controller tried to connect. 

```getsockopt(6, SOL_TCP, TCP_INFO, "\1\0\0\0\0\7w\1P4\3\0@\234\0\0\250\5\0\0\30\2\0\0\0\0\0\0\0\0\0\0"..., [104]) = 0
ioctl(6, FIONREAD, [1])                 = 0
recv(6, "\n", 1, MSG_DONTWAIT)          = 1
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3503, ...}) = 0
write(4, "Sep 05 09:03:46 DEBUG GWT:RFC:C="..., 51) = 51
write(2, "Sep 05 09:03:46 \33[36mDEBUG\33[0m ", 31Sep 05 09:03:46 DEBUG ) = 31
write(2, "GWT:RFC:C=0,MSG=0;255;3;0;2;\n", 29GWT:RFC:C=0,MSG=0;255;3;0;2;
) = 29
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault```

recipedude, 2019-09-05:
Notice quite a few warnings during build.

```hal/architecture/Linux/drivers/core/GPIO.cpp: In constructor ‘GPIOClass::GPIOClass()’:
hal/architecture/Linux/drivers/core/GPIO.cpp:53:18: warning: ‘%s’ directive writing up to 255 bytes into a region of size 48 [-Wformat-overflow=]
    sprintf(file, "/sys/class/gpio/%s/base", de->d_name);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:53:11: note: ‘sprintf’ output between 22 and 277 bytes into a destination of size 64
    sprintf(file, "/sys/class/gpio/%s/base", de->d_name);
    ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:62:18: warning: ‘%s’ directive writing up to 255 bytes into a region of size 48 [-Wformat-overflow=]
    sprintf(file, "/sys/class/gpio/%s/ngpio", de->d_name);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
hal/architecture/Linux/drivers/core/GPIO.cpp:62:11: note: ‘sprintf’ output between 23 and 278 bytes into a destination of size 64
    sprintf(file, "/sys/class/gpio/%s/ngpio", de->d_name);
    ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~```
and 
```In file included from ./MySensors.h:431,
                 from examples_linux/mysgw.cpp:82:
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘bool transportSendWrite(uint8_t, MyMessage&)’ at ./core/MyTransport.cpp:987:2,
    inlined from ‘bool transportRouteMessage(MyMessage&)’ at ./core/MyTransport.cpp:560:40,
    inlined from ‘void stUplinkTransition()’ at ./core/MyTransport.cpp:252:29:
./core/MyMessage.cpp:284:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncation]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘char* protocolMyMessage2Serial(MyMessage&)’ at ./core/MyProtocol.cpp:88:8:
./core/MyMessage.cpp:284:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncation]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function ‘char* MyMessage::getString(char*) const’,
    inlined from ‘char* protocolMyMessage2Serial(MyMessage&)’ at ./core/MyProtocol.cpp:88:8,
    inlined from ‘bool gatewayTransportSend(MyMessage&)’ at ./core/MyGatewayTransportEthernet.cpp:251:51:
./core/MyMessage.cpp:284:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncation]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./core/MyMessage.cpp: In member function ‘char* MyMessage::getString(char*) const’:
./core/MyMessage.cpp:284:17: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying between 0 and 25 bytes from a string of length 25 [-Wstringop-truncation]
    (void)strncpy(buffer, this->data, this->getLength());
          ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘MyMessage& MyMessage::set(const char*)’,
    inlined from ‘void stParentTransition()’ at ./core/MyTransport.cpp:161:29:
./core/MyMessage.cpp:391:15: warning: ‘char* strncpy(char*, const char*, size_t)’ destination unchanged after copying no bytes [-Wstringop-truncation]
  (void)strncpy(this->data, value, this->getLength());
        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘MyMessage& MyMessage::set(const char*)’,
    inlined from ‘void stReadyUpdate()’ at ./core/MyTransport.cpp:312:30:
./core/MyMessage.cpp:391:15: warning: ‘char* strncpy(char*, const char*, size_t)’ destination unchanged after copying no bytes [-Wstringop-truncation]
  (void)strncpy(this->data, value, this->getLength());
        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘MyMessage& MyMessage::set(const char*)’,
    inlined from ‘void stIDTransition()’ at ./core/MyTransport.cpp:218:30,
    inlined from ‘void stIDTransition()’ at ./core/MyTransport.cpp:202:6:
./core/MyMessage.cpp:391:15: warning: ‘char* strncpy(char*, const char*, size_t)’ destination unchanged after copying no bytes [-Wstringop-truncation]
  (void)strncpy(this->data, value, this->getLength());
        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In member function ‘MyMessage& MyMessage::set(const char*)’,
    inlined from ‘bool present(uint8_t, mysensors_sensor_t, const char*, bool)’ at ./core/MySensorsCore.cpp:379:19,
    inlined from ‘void presentNode()’ at ./core/MySensorsCore.cpp:233:15:
./core/MyMessage.cpp:391:15: warning: ‘char* strncpy(char*, const char*, size_t)’ output truncated before terminating nul copying 10 bytes from a string of the same length [-Wstringop-truncation]
  (void)strncpy(this->data, value, this->getLength());
        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘MyMessage& MyMessage::set(const char*)’,
    inlined from ‘void prepareSigningPresentation(MyMessage&, uint8_t)’ at ./core/MySigning.cpp:389:90,
    inlined from ‘bool signerInternalProcessPresentation(MyMessage&)’ at ./core/MySigning.cpp:488:28,
    inlined from ‘bool signerProcessInternal(MyMessage&)’ at ./core/MySigning.cpp:176:42,
[... cut ...]
    inlined from ‘void _process()’ at ./core/MySensorsCore.cpp:72:25,
    inlined from ‘void wait(uint32_t)’ at ./core/MySensorsCore.cpp:566:11,
    inlined from ‘void transportReportRoutingTable()’ at ./core/MyTransport.cpp:1077:8:
./core/MyMessage.cpp:391:15: warning: ‘char* strncpy(char*, const char*, size_t)’ output truncated before terminating nul copying 10 bytes from a string of the same length [-Wstringop-truncation]
  (void)strncpy(this->data, value, this->getLength());
        ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~```  

no idea of it's related.

PR #1346 (2019-09-24): Fix segmentation fault
Fixes #1342
closed 2019-09-24 by mfalkvidd