mysensors/MySensors #1354: #define MY_SMART_SLEEP_WAIT_DURATION being ignored in beta

waspie, 2019-10-16:
I use smartSleep in some sketches and I test at times on a gen 1 rpi that's quite slow and so while testing i extend the smartSleep duration.

In the latest beta, downloaded today ( October 16 2019), mysensors seems to be ignoring this define.
I placed it before and after the Mysensors include just to be safe. This works fine on older releases.

In the current beta it seems to be 500ms no matter the setting.

tekka007, 2019-10-16:
The correct define is `MY_SMART_SLEEP_WAIT_DURATION_MS`, please see the [documentation](https://www.mysensors.org/apidocs-beta/group__SleepSettingGrpPub.html#gaf353bf3627ebd8f15e200b320e2e065d) for reference. However, there is an inconsistency in the documentation that will be addressed in an upcoming PR.

waspie, 2019-10-16:
My mistake. I usually have it right and somehow left that off. Sorry!

PR #1355 (2019-10-23): Code maintenance
- Update Logparser to include transport HAL debug messages (THA:)
- Fix documentation inconsistency, fixes #1354 
- Update keywords
closed 2019-10-23 by tekka007