mysensors/MySensors #1363: Feature request: Speed and course for GPS sensor

ttytyper, 2019-11-08:
Currently the serial protocol supports latitude, longitude and altitude as introduced in #246. That's fine for stationary objects.

But for moving objects it would be nice to also have speed and course. This is particularly useful for dead reckoning. For an example of dead reckoning, check out [aprs.fi](https://aprs.fi/) which often shows moving vehicles with a blue line in front of them, indicating the predicted position based on the last known position, speed and course.

Would it be possible to include these fields in a future version of MySensors?

mfalkvidd, 2019-11-08:
Not sure how controllers handle them, but perhaps V_WIND and V_DIRECTION can be used?

ttytyper, 2019-11-13:
Interesting idea. V_DIRECTION certainly sounds like a good option. V_WIND on the other hand, I'm not so sure. Aside from having the wrong name, it doesn't seem to have a defined unit? Perhaps it would be better to make a new value type, such as V_SPEED?

mfalkvidd, 2019-11-13:
Yes, perhaps it would be. A new value type requires changes to the MySensors library and to all controllers though. Using V_WIND might work with existing versions.

ttytyper, 2019-11-13:
Understood. I'll use V_WIND or a custom type for now.

Do you think something like V_SPEED could be added in a future version or would it require too much change to be practical?

mfalkvidd, 2019-11-13:
For MySensors it is mainly a matter of defining what it should be used for, including thinking if there are other use cases that are related, and whether it needs some other settings (is unit implied from the metric setting, or how is the unit defined, etc). I guess we could see mph, km/h, m/s and maybe others?

Then it would be up to the ~30 different controllers to decide if they want to add support, and how they would like to add that support.