mysensors/MySensors #1371: NRF51822 hangs with heavy traffic and smartsleep

slorenm, 2019-12-07:
I've noticed using smartsleep with nrf51 platform, when heavy traffic is applied to the system (i.e. a lot of messages during switching groups), nrf51 hangs in Timer interrupt 'NRF5_RADIO_TIMER_IRQ_HANDLER' of Radio_ESB.cpp, with never cleared status of 'NRF5_RADIO_TIMER->EVENTS_COMPARE[1]'. This results in interrupt being handled over and over again i blocks execution. What I found in the code, this status should be handled using PPI channel. Clearing this status in debug resolves the issue until next such case occurs.
I found a solution in 'static bool NRF5_ESB_initialize()' changing radio and timer intterrupt priorities (basically swapping them) .With this, problem never occurs anymore.
Could someone take a look at this and verify the solution?

petewill, 2019-12-31:
@slorenm I believe I am having this same problem.  Can you please be more specific about what exactly you changed in the code?  Are you saying you set
`NVIC_SetPriority(RADIO_IRQn, 1);` to `NVIC_SetPriority(RADIO_IRQn, 2);`
and
`NVIC_SetPriority(NRF5_RADIO_TIMER_IRQN, 2);` to `NVIC_SetPriority(NRF5_RADIO_TIMER_IRQN, 1);`
in Radio_ESB.cpp?
Thanks,
Pete

slorenm, 2020-01-01:
Yes so basically timer interrupt has higher priority than the radio with the change. It's kind of major change so it would be nice to confirm it's a real fix rather than just a cover up for some other issue.
Btw. sorry for my previous blurry explanation

petewill, 2020-01-01:
@slorenm Thanks!  I just made the change and I'm testing it now.  In my experience the node will run fine for a few days then crash so I'll report back in a few days with my results.

petewill, 2020-01-29:
@slorenm Thank you for this change.  My device has been working flawlessly for a few weeks since I made the change.

fvpetrov, 2020-04-28:
@slorenm 
I took a look at the code last night, due to my #1410 .

For me, the code looks wrong.

`EVENTS[1]` seems to be used to power off the radio, when no ACK is received after waiting `NRF5_ESB_ACK_WAIT` microseconds.
This feature is only activated when `NRF5_ESB_USE_PREDEFINED_PPI` is defined.
In this case, the radio will be powered down after ACK's timeout until retransmission, saving some power,
otherwise the radio will actually wait for an ACK until retransmission.

`EVENTS[3]` is the retransmission timer, obvious from timer's  interrupt handler.

For this, to work as expected:
 - timer's interrupt has to be fired by `EVENTS[3]` (line L#195),
 - `EVENTS[3]` has to be used within interrupt's handler (L#700).

Reference:
https://github.com/mysensors/MySensors/blob/8103c6cf6bcb3e271df08b17d0fb2fee3b444c86/hal/transport/NRF5_ESB/driver/Radio_ESB.cpp#L195

https://github.com/mysensors/MySensors/blob/8103c6cf6bcb3e271df08b17d0fb2fee3b444c86/hal/transport/NRF5_ESB/driver/Radio_ESB.cpp#L700

nikolac, 2020-09-08:
I'm having the same issue with both my NRF51822 & NRF52832 nodes. At some point, they freeze, consume huge amounts of power. Sometimes triggering brownout which brings them back. Other nodes require manual intervention.

phodgers, 2020-09-27:
Check out my post in similar nRF52 thread about deadlocking https://github.com/mysensors/MySensors/issues/1410#issuecomment-699647081 I think maybe the issue is due to oscillator error? Maybe you can test to see if it is the same reason for you?

d00616, 2020-10-18:
There is a fix for the ESB radio https://github.com/mysensors/MySensors/pull/1445

closed 2020-11-07 by mfalkvidd

rikki78, 2022-03-03:
I also experienced the NRF to hang after sending. The fix as described by @slorenm , solved this as far as I see now. 
However when I tried to use the development branch (to include the deadlock fix as mentioned by @d00616), the NRF5 doesn't find its parent. 
What is the right fix?