mysensors/MySensors #1376: Gateway ESP32 DevKit V1 & MQTT & RF24 unable connect to WIFI

eduardomarcos, 2020-01-01:
Hardware using:
- ESP32 DevKit V1
- RF24l01+
Both components works (previously checked with other programs)

I tried to install in the ESP32 the Gateway working in Ethernet mode. It works with current wiring (I followed the setup instructions)

But, when I try to configure it to work as an MQTT Gateway, it is unable to connect to Wifi.
MQTT service is a mosquitto running locally (and it works good)

Serial Output:
**1181 MCO:BGN:STP
1182 MCO:BGN:INIT OK,TSP=1
1190 GWT:TPC:CONNECTING...
2193 TSM:READY:NWD REQ
2198 ?TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
2207 GWT:TPC:CONNECTING...
3213 GWT:TPC:CONNECTING...
4220 GWT:TPC:CONNECTING...
5226 GWT:TPC:CONNECTING...
6232 GWT:TPC:CONNECTING...
7239 GWT:TPC:CONNECTING...
8246 GWT:TPC:CONNECTING...
9253 GWT:TPC:CONNECTING...
10260 GWT:TPC:CONNECTING...
11267 GWT:TPC:CONNECTING...**

And it is always trying to connect :(

And the code that I'm using:
```
#define MY_DEBUG
#define MY_RADIO_RF24
#define MY_WIFI_SSID "ssid"
#define MY_WIFI_PASSWORD "pass"

#define MY_HOSTNAME "ESP32_GW"
#define MY_GATEWAY_MQTT_CLIENT
#define MY_GATEWAY_ESP32

#define MY_MQTT_PUBLISH_TOPIC_PREFIX "mygateway1-out"
#define MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "mygateway1-in"
#define MY_MQTT_CLIENT_ID "mysensors-1"
#define MY_MQTT_USER "mosquitto"
#define MY_MQTT_PASSWORD "mosquitto"

// MQTT broker ip address.
#define MY_CONTROLLER_IP_ADDRESS 127, 0, 0, 1
// The MQTT broker port to to open
#define MY_PORT 8883

#include <MySensors.h>

void setup()
{

}

void presentation()
{
  // Present locally attached sensors here
}

void loop()
{
  // Send locally attached sensors data here
}
```

I tried different configurations and I always get the same result. Not possible to connect (But Ethernet Gateway mode with same hardware wiring works)

Any idea? Maybe I'm missing something?

Thanks in advance!

eduardomarcos, 2020-01-01:
Well, after continuing with some research, I found out a two things.

1. I do not know how I was thinking about connecting to a localhost MQTT server from the ESP32... Silly me! So I changed the MQTT server to a reachable one, that also works.

2. I force the WiFi connection in the setup function. And it also worked. After doing that, I could connect to WiFi, and the rest of the configuration was successful, brilliant.

The output that I got:

1183 MCO:BGN:STP
Force WiFi Connection

[Connecting] [-----------[IP address: 192.168.1.38]
6689 MCO:BGN:INIT OK,TSP=1
6694 GWT:TPC:IP=192.168.1.38
6698 GWT:RMQ:CONNECTING...
7197 GWT:RMQ:OK
7199 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/18,MSG SENT
7209 TSM:READY:NWD REQ
7214 ?TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:

And the code that I added:
```
void setup()
{
  Serial.begin(115200);
  Serial.println("Force WiFi Connection");
  WiFi.begin(MY_WIFI_SSID, MY_WIFI_PASSWORD);
    Serial.print("\n[Connecting] [");
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print("-");
    }

    Serial.print("[IP address: ");
    Serial.print(WiFi.localIP());
    Serial.print("]\n");

}
```

milvert, 2021-04-30:
When I added the timeout to 3 seconds it works.

```
bool gatewayTransportConnect(void)
{
#if defined(MY_GATEWAY_ESP8266) || defined(MY_GATEWAY_ESP32)
	if (WiFi.status() != WL_CONNECTED) {
		GATEWAY_DEBUG(PSTR("GWT:TPC:CONNECTING...\n"));
		delay(3000); // Changed from delay(1000);
		return false;
	}
	GATEWAY_DEBUG(PSTR("GWT:TPC:IP=%s\n"), WiFi.localIP().toString().c_str());
```

Should I create a pull request with this change?

denisjoshua, 2021-04-30:
I also have same ESP32 DevKit V1 and I had same problem... 
I added the timeout to 3 in the gatewayTransportConnect and it's work. 
All this if I use the official release. 
But if I use the DEV release VER=2.4.0-alpha is not work... it give me error: 
```
Stack smashing protect failure!
abort() was called at PC 0x40134cef on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x40088620:0x3ffb1eb0 0x4008889d:0x3ffb1ed0 0x40134cef:0x3ffb1ef0 0x400d27e6:0x3ffb1f10 0x400d281e:0x3ffb1f50 0x400d2a23:0x3ffb1f70 0x400d2ff4:0x3ffb1f90 0x400d3105:0x3ffb1fb0 0x400898ae:0x3ffb1fd0
Rebooting...
```

Thanks all
Denis

zetxx, 2022-03-05:
> gatewayTransportConnect

+1