mysensors/MySensors #1379: Clarify CLA

herm, 2020-01-05:
The CLA states

> 2.3 Outbound License
> As a condition on the grant of rights in Sections 2.1 and 2.2, We agree to license the Contribution only under the terms of the license or licenses which We are using on the Submission Date for the Material (including any rights to adopt any future version of a license if permitted). 

[Here](https://www.mysensors.org/about/mysensors#copyright) it looks like you are offering a commercial license but don't give details about the license. 

Could you please clarify which licenses exactly CLA section 2.3 refers to?

mfalkvidd, 2020-01-05:
We've tried (in https://github.com/mysensors/MySensors/issues/1242 ) but there is not yet a solution unfortunately.

Would you mind if we close this issue and keep 1242?

herm, 2020-01-06:
#1242 seems to be more concerned about GPL 2 vs GPL 3. I am more interested in what commercial licenses you are considering valid under the CLA. But you can track both issues in #1242.

closed 2020-01-06 by herm