mysensors/MySensors #1382: Add ATtiny Support

fabyte, 2020-01-09:
Hi,
I want to contribute my changes to MySensors in order to support ATtiny MCUs (see the [project page](https://forum.mysensors.org/topic/10866/a-tiny-bme-node-bme280-on-attiny85)).

Three issues had to be addressed:
1. Make RF24 driver be able to handle CE pin pulled up to VCC (already created PR #1381 for this)
2. Some register names had to be redefined and a function had to be defined to be able to compile MySensors for ATtiny.
3. Fix hwCPUFrequency() in MyHwAVR.cpp

I want to discuss these issues:

2. The missing defines are:
```
#define SIGRD   RSIG    // Signature Row from Store Program Memory Control and Status Register
#define EIFR    GIFR    // External/General Interrupt Flag Register
#define WDTCSR  WDTCR   // WDT Control Register
// #define TCCR1A TCCR1 // optional
// #define TIFR1 TIFR	// optional
extern void serialEventRun(void) __attribute__((weak));
```
Where should these definitions be places? Is there some kind of compatiblity.h header in the project for different MCUs?

3. The hwCPUFrequency() seems to dynamically measure the CPU frequency and for this all timers are suspended. Since ATtiny supports not that much timers, the more simple approach for a fix would be like for other hardware platforms:
```
@@ -309,6 +309,11 @@ uint16_t hwCPUVoltage(void)
 uint16_t hwCPUFrequency(void)
 {
+#if defined (__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__) || \
+       defined (__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
+       // reporting compile time frequency (in 1/10MHz)
+       return F_CPU / 100000UL;
+#else
        cli();
        // save WDT & timer settings
        const uint8_t WDTsave = WDTCSR;
@@ -343,6 +348,7 @@ uint16_t hwCPUFrequency(void)
        TCCR1C = TCCR1Csave;
        // return frequency in 1/10MHz (accuracy +- 10%)
        return result;
+#endif
 }
```

What do you think?

chamroeunbm, 2021-04-13:
More than 1 years since your post but noone replies or pay attention to it. How is it going at your end regarding this. I am also trying to get Attiny167 to work with Mysensors but failed. Even though attiny is small(8KB) but it's enough for many single sensor device.

fabyte, 2021-04-14:
Unfortunately there is not much support from the project. You can try and use my [attiny](https://github.com/fabyte/MySensors/tree/attiny) branch. It contains changes to address number 1. and 3. of this issue.

chamroeunbm, 2021-04-19:
thanks, your approach works. now I got my Attiny167+Sht30 working with Mysensors and Mysensor gateway. Next step will be formatting the data for Thingsboard cloud.