mysensors/MySensors #1389: Truncated message when arduino sending version info

damianchojna, 2020-02-09:
I check that in mMySensors arduino lib 2.3.2 on ArduinoMega, when i use node as MY_GATEWAY_SERIAL

Problem with the cut message when arduino send version info:
`0;255;0;0;17;2 /*!< Major release vers`

In Arduino/libraries/MySensors/core/Version.h, in MYSENSORS_LIBRARY_VERSION_MINOR is this truncated message. This makes mysensors in the home assistant not work.

Line 58:
`
#define MYSENSORS_LIBRARY_VERSION STR(MYSENSORS_LIBRARY_VERSION_MAJOR) "." STR(MYSENSORS_LIBRARY_VERSION_MINOR) "." STR(MYSENSORS_LIBRARY_VERSION_PATCH) //!< final release versioning`
When I replace it with, this work for me.
`MYSENSORS_LIBRARY_VERSION "2.3.2"`

This truncated information is correct behavior?

Arduino Code:
```
#define MY_GATEWAY_SERIAL
#define MY_REPEATER_FEATURE
#define MY_NODE_ID 0
#include <SPI.h>
#include <MySensors.h>
#include <Bounce2.h>

#define RELAY_PIN  5
#define BUTTON_PIN  A0
#define CHILD_ID 1
#define RELAY_ON 1
#define RELAY_OFF 0

Bounce debouncer = Bounce();
bool state = false;
bool initialValueSent = false;

MyMessage msg(CHILD_ID, V_STATUS);

void setup()
{
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  debouncer.attach(BUTTON_PIN);
  debouncer.interval(10);

  // Make sure relays are off when starting up
  digitalWrite(RELAY_PIN, RELAY_OFF);
  pinMode(RELAY_PIN, OUTPUT);
}

void presentation()  {
  sendSketchInfo("Relay+button", "1.0");
  present(CHILD_ID, S_BINARY);
}

void loop()
{
  if (!initialValueSent) {
    Serial.println("Sending initial value");
    send(msg.set(state?RELAY_ON:RELAY_OFF));
    Serial.println("Requesting initial value from controller");
    request(CHILD_ID, V_STATUS);
    wait(2000, C_SET, V_STATUS);
  }
  if (debouncer.update()) {
    if (debouncer.read()==LOW) {
      state = !state;
      // Send new state and request ack back
      send(msg.set(state?RELAY_ON:RELAY_OFF), true);
    }
  }
}

void receive(const MyMessage &message) {
  if (message.isAck()) {
     Serial.println("This is an ack from gateway");
  }

  if (message.type == V_STATUS) {
    if (!initialValueSent) {
      Serial.println("Receiving initial value from controller");
      initialValueSent = true;
    }
    // Change relay state
    state = (bool)message.getInt();
    digitalWrite(RELAY_PIN, state?RELAY_ON:RELAY_OFF);
    send(msg.set(state?RELAY_ON:RELAY_OFF));
  }
}
```

Serial Output:
```
0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2 /*!< Major release vers
0;255;3;0;11;Relay+button
0;255;3;0;12;1.0
0;1;0;0;3;
```

mfalkvidd, 2020-02-09:
Interesting. That line of code hasn't been changed since Sep 4, 2017.

mfalkvidd, 2020-02-09:
Cross-reference: https://forum.mysensors.org/topic/10934/truncated-message-when-arduino-sending-version-info

niclas0219, 2020-06-11:
@mfalkvidd Wow.. Nice clarification! I was sure that the gateway had a list of nodes stored in EEPROM and was 100% in charge of assigning NODE IDs. I'll have to remember that in the future... and read some more..