mysensors/MySensors #1407: Raspberry Gateway : Add variable in conf to modify service name

3daerovision, 2020-04-19:
Hi,
Would it be possible to add a parameter in the configuration to change the name of the generated binairy / service ?
Thanks :)

mfalkvidd, 2020-04-21:
Thanks for suggesting this.

Could you share the main reason for wanting to rename the binary and service? Understanding the main reason can be good so we don't miss some small but important aspect related to the rename.

3daerovision, 2020-04-22:
I run 2 instances of the service on my Raspberry, for 2 transport mode. Changing the name in the config would make this easier. But it could be usefull if someone wan't to test different setting without replacing his working instance of the service.

mfalkvidd, 2020-05-08:
I have started work on this. Service naming works, but to make it possible to run more than 1 instance we need to separate the configuration file, eeprom file, log file, log pipe and maybe more.

mfalkvidd, 2020-05-13:
I'm worried that people will be confused when they compile and run two gateways with different service names, but same config file, log file, etc. I have not found a good way to handle this. Suggestions are welcome.

3daerovision, 2020-05-13:
I understand. I was not thinking of having multiple gateways on one config file, but more like multiple directories, with as many config and log files. For my tests, I have 2 separate MYS directories, have build both with different config, and then I had to rename the executable file and in services. I can run both without problem.
So, maybe just add the name for service in config file is enough ?

mfalkvidd, 2020-05-14:
For brand new installations, we could add the name of the service as a setting in the config file, and let the gateway check the name at startup and refuse to start if the name doesn't match. But then we'd need a safe way to handle modification of the config file when people upgrade.

mfalkvidd, 2020-05-14:
Here is what I'm thinking so far:

- Add the service name to each log message. That would make it possible to read the log file even if multiple instances use the same log file/pipe.
- At startup, log the full file name and path for eeprom file and config file
- Create a file lock on the eeprom file, and fail startup if the gateway cannot get the lock.

I'm not sure these are the best options, or if they will be sufficient to keep most people out of trouble.

mfalkvidd, 2020-05-14:
New updates in https://github.com/mfalkvidd/MySensors/tree/rpi-servicename

Unfortunately, I am not skilled enough to figure out how to lock the eeprom file.

mfalkvidd, 2020-05-16:
@3daerovision when you have time, could you please test https://github.com/mysensors/MySensors/pull/1421 ?

mfalkvidd, 2020-05-18:
I did some work trying to lock the eeprom file last weekend, thanks to hints from Yveaux. Judging by the progress so far, I'll need multiple more weekends to figure out how to do it.

mfalkvidd, 2020-06-24:
@3daerovision have you had any opportunity to test #1421 ?

MySensors-MJChrist, 2020-08-04:
Hi mfalkvidd,

Is your work in the master or the development branch?
I want to do the same on a raspberry with wireless and rs484 gateway. Can i also test for you.
Do you think it is possible to use two wireless by using the second spi bus on the raspberry?

mfalkvidd, 2020-08-04:
The work is on the development branch.

There are now two ways to use a radio and rs485.
1. Use the new support for multiple transports in a single gateway. See https://forum.mysensors.org/topic/11135/something-s-cooking-in-the-mysensors-labs
2. Use two instances of the raspberry pi gateway (on the same raspberry pi), with or without https://github.com/mysensors/MySensors/pull/1421 See https://forum.mysensors.org/topic/7702/double-spi-radio-raspberry-pi

MySensors-MJChrist, 2020-08-04:
If i want to make two instance of the gateway on the Raspberry i get te following:
[WARNING] Unknown option detected:--service-name=mys1gw, ignored

This is the configure command:
./configure --service-name=mys1gw --my-transport=rf24 --my-rf24-channel=83 --my-rf24-pa-level=RF24_PA_HIGH --my-rf24-ce-pin=22 --my-rf24-cs-pin=24 --my-transport=rf24 --my-rf24-irq-pin=15 --my-gateway=ethernet --my-port=5003 --my-leds-err-pin=12 --my-leds-rx-pin=16 --my-leds-tx-pin=18

Its not very clear how to make the two instance of the gateway.

For cloning i used:
git clone https://github.com/mysensors/MySensors.git --branch development

mfalkvidd, 2020-08-04:
https://github.com/mysensors/MySensors/pull/1421 has not been merged yet, so the service name feature is not available there.