mysensors/MySensors #1416: Problem with DFR0342 / W5500

grzana12, 2020-05-10:
I bought this: https://wiki.dfrobot.com/W5500_Ethernet_with_POE_Mainboard_SKU__DFR0342

I am trying to integrate with MySensor library. I found information that I should use MY_GATEWAY_ENC28J60.
PoE works fine, but on Serial output I see:

> 20016 TSM:FAIL:RE-INIT
> 20016 TSM:INIT
> 20021 !TSM:INIT:TSP FAIL
> 20022 TSM:FAIL:CNT=3
> 20022 TSM:FAIL:DIS
> 20022 TSF:TDI:TSL
> 30024 TSM:FAIL:RE-INIT
> 30024 TSM:INIT
> 30029 !TSM:INIT:TSP FAIL
> 30030 TSM:FAIL:CNT=4
> 30030 TSM:FAIL:DIS
> 30030 TSF:TDI:TSL
> 40032 TSM:FAIL:RE-INIT
> 40032 TSM:INIT
> 40037 !TSM:INIT:TSP FAIL
> 40038 TSM:FAIL:CNT=5
> 40038 TSM:FAIL:DIS
> 40038 TSF:TDI:TSL

Any idea?

mfalkvidd, 2020-05-10:
I don't think ENC_28J60 is compatible with W5500. Maybe the instructions at https://www.mysensors.org/build/ethernet_gateway are more applicable.

Also see https://forum.mysensors.org/topic/666/read-this-first-it-could-save-you-a-lot-of-time/ if you haven't already. It contains the most common problems and how to troubleshoot them efficiently.

About the messages you posted, they indicate that the gateway is unable to initialize whatever transport you have selected. Which transport are you using?

grzana12, 2020-05-10:
Problem solved. Gateway ENC28J60 works for W5500. Problem was in code I forgot remove radio definition

closed 2020-05-10 by grzana12