mysensors/MySensors #1420: Intercepting network messages in a gateway

SinishaDjukic, 2020-05-14:
During development it useful to have a way to monitor sensor messages on a gateway. For example, I would like to monitor the low power sensor battery levels over a period of days and weeks to ensure that low power has really been achieved.

For sure, this can be done with some of the existing controller UIs, but this also has certain disadvantages. E.g. you need to have a controller machine up and running for the whole time, or set up a dedicated RPi just for the purpose of experimentation and development.

Instead I'm using M5Stack and displaying the latest message (sender, type, value) and the latest known sensor state (battery, temperature, etc.). This is very convenient and mobile (built-in battery), and easy to modify and extend.

In order to do so I've modified gatewayTransportSend in MyGatewayTransportSerial.cpp and added a hook there. But is there a better way?

mfalkvidd, 2020-05-14:
Can't you use the official API (the same that all controllers use)? https://www.mysensors.org/download/serial_api_20

SinishaDjukic, 2020-05-14:
Using the serial gateway implementation on the one side and implementing a serial controller on the other - all on the same device - seems a bit of an overkill...

The current MY_GATEWAY_XXX defines instantiate specific gateway implementations (serial, MQTT, etc.), but would it work if do not use any of the default implementations? Like this:
1) define these instead:
#define MY_GATEWAY_FEATURE
#define MY_IS_GATEWAY (true)
#define MY_NODE_TYPE "GW"
2) implement the functions from MyGatewayTransport.h

Anything missing?

TRSx80, 2020-06-29:
Not sure how you progress on this, and maybe my suggestion will be more "heavy weight" than you would like, but personally I use MQTT as something in between a full controller installation and just the gateway.  Yes it will require some SBC on your network, however for most of us hacking on these sort of things that is not a problem (I know that I personally have several such machines running anyway, for instance).  `mosquitto` broker is only an `apt install` away in Debian, and then with tools like [`mqttwarn`](https://github.com/jpmens/mqttwarn) you can log this stuff directly into databases (or text files, if you prefer) right at the MQTT level, with no controller required.  This is what I plan on doing, and am about halfway along in implementing it.