mysensors/MySensors #1424: Adafruit nRF52840 Feather support

Avamander, 2020-05-30:
I read on the forums and from the documentation that the nRF5 series are supported to some extent. I am wondering what does it actually take to get the gateway sketch running on the device? 

Right now it didn't even detect the board properly so I had to help it a bit - I changed Flash.h to load the NRF5 HAL and nRF52840 flash parameters. 

But now it's just erroring out further:
```
In file included from /home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/architecture/NRF5/MyHwNRF5.h:51:0,
                 from /home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/architecture/NRF5/MyHwNRF5.cpp:23,
                 from /home/a/Programming/Workspace/Arduino/libraries/MySensors/MySensors.h:74,
                 from /tmp/arduino_modified_sketch_572985/GatewaySerial.ino:89:
/home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/architecture/NRF5/drivers/nrf5_wiring_digital.c: In function 'void nrf5_pinMode(uint32_t, uint32_t)':
/home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/architecture/NRF5/drivers/nrf5_wiring_digital.c:37:10: error: 'g_APinDescription' was not declared in this scope
  ulPin = g_APinDescription[ulPin].ulPin;
          ^~~~~~~~~~~~~~~~~
In file included from /home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/transport/NRF5_ESB/driver/Radio.cpp:1:0,
                 from /home/a/Programming/Workspace/Arduino/libraries/MySensors/MySensors.h:364,
                 from /tmp/arduino_modified_sketch_572985/GatewaySerial.ino:89:
/home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/transport/NRF5_ESB/driver/Radio.h: At global scope:
/home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/transport/NRF5_ESB/driver/Radio.h:68:17: error: 'RADIO_MODE_MODE_Nrf_250Kbit' was not declared in this scope
  NRF5_250KBPS = RADIO_MODE_MODE_Nrf_250Kbit, // Deprecated!!!
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/a/Programming/Workspace/Arduino/libraries/MySensors/hal/transport/NRF5_ESB/driver/Radio.h:68:17: note: suggested alternative: 'RADIO_MODE_MODE_Nrf_2Mbit'
  NRF5_250KBPS = RADIO_MODE_MODE_Nrf_250Kbit, // Deprecated!!!
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                 RADIO_MODE_MODE_Nrf_2Mbit
```

Compilation flags used by Arduino IDE:
```
-mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=64000000 -DARDUINO=10813 -DARDUINO_NRF52840_FEATHER -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"0.20.1\"" -DNRF52840_XXAA -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8029 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Feather nRF52840 Express\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -DCFG_LOGGER=1 -DCFG_SYSVIEW=0 -
```

phodgers, 2021-03-12:
I don't think the nRf52840 supports 250KPS speed, so you need to set to 1Mbit and also remove reference to the 250KBPS in the config. Alo the defintiion for pins is only up to 32 pins i.e. P0.0-P0.31. the adafruit uses connector P1.0-P1.31. So the pin defintions in MySensors needs updated to cover up to 64 pins i.e. pin references 0-63. I put a post about this here in MySensors forum [https://forum.mysensors.org/topic/11590/nrf52840-support-on-mysensors?_=1615562114364]

milvert, 2021-06-06:
@Avamander Will https://github.com/mysensors/MySensors/pull/1473 solve this issue?

Avamander, 2021-06-06:
> @Avamander Will https://github.com/mysensors/MySensors/pull/1473 solve this issue?

Not sure, it might not if the references remain (even if not used).

I'd love to have it working, but I don't know how exactly.