mysensors/MySensors #1430: Compile error building with updated ESP8266 boards manager release

AlphaSierraHotel, 2020-06-07:
Compiling the MySensors GatewayESP8266MQTTClient example script as-is.  I'm getting the following errors during the linking stage.

```
Linking everything together...
"C:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800/local.eagle.app.v6.common.ld"
"C:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -fno-exceptions -DNEW_OOM_ABORT -Wl,-Map "-Wl,C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800/GatewayESP8266MQTTClient.ino.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/lib" "-LC:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/lib/NONOSDK22x_191024" "-LC:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/ld" "-LC:\\Users\\ahubble\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/libc/xtensa-lx106-elf/lib" -Teagle.flash.4m2m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800/GatewayESP8266MQTTClient.ino.elf" -Wl,--start-group "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800\\sketch\\GatewayESP8266MQTTClient.ino.cpp.o" "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800\\libraries\\ESP8266WiFi\\ESP8266WiFi.a" "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800\\libraries\\MySensors\\MyASM.S.o" "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800\\libraries\\EEPROM\\EEPROM.cpp.o" "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_cache_367522\\core\\core_6c209f3ca594be3a1fb4b73fc3684dc9.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\\Users\\ahubble\\AppData\\Local\\Temp\\arduino_build_299800"
c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o):C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:55: multiple definition of `g_pcont'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:52: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `esp_yield':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:103: multiple definition of `esp_yield'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:88: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `esp_schedule':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:111: multiple definition of `esp_schedule'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:96: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `__yield':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:114: multiple definition of `__yield'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:101: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `optimistic_yield':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:126: multiple definition of `optimistic_yield'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:112: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `init_done()':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:232: multiple definition of `init_done()'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:152: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function `app_entry':

C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:300: multiple definition of `app_entry'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:205: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o):C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\cores\esp8266/core_esp8266_main.cpp:49: multiple definition of `resetInfo'; C:\Users\ahubble\AppData\Local\Temp\arduino_build_299800\sketch\GatewayESP8266MQTTClient.ino.cpp.o:C:\Users\ahubble\Dropbox\Tech\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:46: first defined here

c:/users/ahubble/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ahubble\AppData\Local\Temp\arduino_cache_367522\core\core_6c209f3ca594be3a1fb4b73fc3684dc9.a(core_esp8266_main.cpp.o): in function ` [... cut ...]
Using library EEPROM at version 1.0 in folder: C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\EEPROM 
Using library SPI at version 1.0 in folder: C:\Users\ahubble\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\libraries\SPI 
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
```

This seems to start with release 2.6.0 of [esp8266/Arduino boards manager package](https://github.com/esp8266/Arduino).  Falling back to v2.5.2 compiles fine.  I've also tested with 2.7.1 (current), 2.5.1 and 2.5.0.  Before 2.6.0 it compiles completely

mfalkvidd, 2020-06-08:
MySensors 2.3.2 was updated to work with esp8266 board 2.6.2, so the solution is probably to upgrade MySensors. See section "Important information" at https://github.com/mysensors/MySensors/releases/tag/2.3.2

AlphaSierraHotel, 2020-06-08:
Confirmed.  MySensors 2.3.2 compiles with esp8266 board >= 2.6.2.

Thanks.

closed 2020-06-08 by AlphaSierraHotel