mysensors/MySensors #1456: RF24 reports that 50% TX Powerlevel is used with RF24_PA_MAX

joro75, 2020-11-26:
When MY_SIGNAL_REPORT_ENABLED is defined, the Transmit power percentage (SR_TX_POWER_PERCENT) can be requested.
When the powerlevel of the RF24 is set to RF24_PA_MAX (which is the maximum), 50% is returned.
This is not correct, and also doesn't match with the Power level that can be requested (SR_TX_POWER_LEVEL), which is correctly reported as 0 dBm (which is the maximum).

This is caused by a wrong shift of the RF_SETUP register on line 412 of RF24.cpp (https://github.com/mysensors/MySensors/blob/8f1f8ad2415034569a77d99f6ed9c02573c71418/hal/transport/RF24/driver/RF24.cpp#L412) 
The right shift shouldn't be 2 but it should be a right shift of 1, as it is used on line 406.

tekka007, 2020-11-26:
@joro75 Good catch! Thanks for reporting

joro75, 2020-11-26:
@tekka007 Not a problem. Thanks for fixing!

PR #1457 (2020-11-28): RF24: Fix wrongly reported TX level
- Fixes #1456
closed 2020-11-28 by tekka007