mysensors/MySensors #1459: RPi RF24 gateway with interrupt pin & repeater node error

acejoh, 2020-12-12:
Hi,

I've run into an issue with my RPi gateway using the interrupt feature (pin 15). As soon as a node starts transmitting through a repeater node I can see "NOTICE RF24: Recovered from a bad interrupt trigger." messages in the gateway log.
I have 2 nodes in my current setup.
Node 1 is a sensor node and repeater node. Messages from node 1 are received without problem by the gateway
Node 2 is a sensor node that has node 1 as a static parent. It is a passive node. Messages from this node are not received by the gateway
Note that if I set node 2's parent as the gateway (0) the messages from node 2 are being received by the gateway.
Also note that if I disable the interrupt pin for the RPi gateway the problem disappears. The gateway now receives messages from node 2 that come through node 1.

If you need more information, please let me know!

Thank you for looking into this issue.

mfalkvidd, 2020-12-12:
Thanks for this report. A full debug log from all involved nodes is usually very useful to figure out what happens, so it would be great if you capture that and post here.

acejoh, 2020-12-14:
Hi,

Unfortunately my repeater node is not easily accessible and I do not have a spare arduino that I can use to replace it with.

I'll attach the gateway and sending node logs (node 2) here:
[node2.log](https://github.com/mysensors/MySensors/files/5690850/node2.log)
[gateway.log](https://github.com/mysensors/MySensors/files/5690851/gateway.log)

this is my configure line for the RPi gateway:
./configure --my-transport=rf24 --my-gateway=ethernet --my-port=5003 --my-rf24-irq-pin=15 --my-leds-err-pin=12 --my-leds-rx-pin=16 --my-leds-tx-pin=18

Also: the source code for Node 1
[Node1.txt](https://github.com/mysensors/MySensors/files/5690871/Node1.txt)

and Node 2
[Node2.txt](https://github.com/mysensors/MySensors/files/5690875/Node2.txt)

mfalkvidd, 2020-12-15:
Thanks. Troubleshooting this is unfortunately beyond my knowledge, but hopefully someone else can pitch in.