mysensors/MySensors #1464: Allow to define the NODE_ID by hardware

frapell, 2021-01-14:
Currently, the library allows to either set the NODE_ID from the gateway automatically, or defining it in the firmware when uploading the code to the chip.

What I propose here is a way to define the NODE_ID using external means, like maybe using jumpers with input pins or something (haven't thought yet what would be the best way)

Te use case is the following, I have several boxes around the house controlling lights and sockets so I can turn them on/off. The issue is that when I add changes to the logic, I need to remember which NODE_ID I assigned to each box so I can define it properly before uploading the code (I don't like auto assigning the node id).

what this change would allow is to fix the node id somehow in the PCB, and simply upload the same code to every arduino, and so when they are turned on, they will have the same node id as before... Also, if for whatever reason I would like to change it, I don't need to re-upload the code to it...

I could try to give it a shot and implement it myself, but I don't know if it is a desired feature or maybe it was already discussed and discarded in the past?

mfalkvidd, 2021-01-17:
People have already done similar solutions. See https://forum.mysensors.org/topic/5410/changing-node-id-at-run-time/ for one way to do it.

frapell, 2021-01-18:
Wow, totally missed that! Thank you!

closed 2021-01-18 by frapell