mysensors/MySensors #1466: Remove plantuml.jar

rstephan, 2021-01-30:
Is there a reason to ship a 7 MB binary blob with every release? I mean, other than laziness. That's 3/4 waste, for 1/4 project. You have an environment variable and a build-script. This should be enough to build the docs from CI without any trouble.
🌍 Save the planet and reduce some traffic!

mfalkvidd, 2021-01-30:
Could you expand on that? The Documentation folder is excluded from all releases ([reference](https://github.com/mysensors/MySensors/blob/development/library.json#L16)), so unless Arduino IDE is broken, the jar file will never be included in any release.

rstephan, 2021-02-01:
Thanks for the reply.

My Arduino-IDE (1.8.12/win) pulled a "MySensors-2.3.2.zip" with 10.2 MB in size. A exact copy of the repository, as far as I can tell at first glance.

The downloads links from your website reference the github master branch (no excludes to see), also 10.2 MB. The headline is "Latest Release". 

What am I doing/seeing wrong?

mfalkvidd, 2021-02-01:
Thanks rstephan. That's interesting. I usually don't use the Arduino IDE Library Manager myself, since I need the full repository for development (including generating the documentation, since I work more with the documentation than with the code). On the other hand, there is no way I would notice if a library pulled down an extra 7MB (my Arduino folder uses a bit more than 600MB) so I might be the right person to discuss this anyway.

If the Arduino IDE doesn't obey the `exclude` directive, then I guess the only option is to remove the jar file from the repository, store it at some other place and modify the doxygen script to detect if the file is missing so it can instruct developers where to download it manually from, and where to save it whenever they work on code locally. Is that something you could prepare a pull request for?

tbowmo, 2021-02-02:
If I'm not totally wrong, our build pipeline generate the zip file, so perhaps we just need to look at that, to exclude unwanted objects in our build artifact.

rstephan, 2021-02-03:
@tbowmo in any case, the pull-request is ready.