mysensors/MySensors #1489: hwCPUVoltage() screws up analogRead() readings

gorbunovav, 2021-05-04:
Simple code to test is looking something like this:

```c++
void setup()
{
    analogReference(INTERNAL);
}

void loop()
{
    analogRead(pin_number); // good value
    hwCPUVoltage();
    analogRead(pin_number); // bad value
}
```

It happens on Arduino Pro Mini 3.3V for me.

There is also an older thread on the forum:
https://forum.mysensors.org/topic/6808/hwcpuvoltage-on-avr-screws-up-analogread-readings?lang=ru

virtual-maker, 2021-05-05:
@gorbunovav 
As I understand the MySensors lib code hwCPUVoltage() changes the voltage reference for the ADC to 1.1V (Vbg)
and after changing the analog reference back (with your second analogRead), the first few readings from analogRead() may not be accurate. Normally there is an external capacitor to AREF pin which needs to align to the new internal AREF voltage.
May be you should add a small loop with some "dummy" analogRead() before you read your actual value. You can check this when you debug print out the read "dummy" values in the loop.

gorbunovav, 2021-05-29:
@virtual-maker thank you! I've tried - dummy loop works

closed 2021-05-29 by gorbunovav