mysensors/MySensors #1495: ESP8266 Gateway always crashes

Zwer2k, 2021-08-08:
Can't get ESP8266 gateway to work anymore

Hello, 

I have been using MySensors for a long time. First with serial gateway now with ESP8266 gateway. For months everything worked fine except for small dropouts. Due to dropouts I wanted to switch to current development branch, since then I can not get the gateway to work.
The ESP delivers only the following output:
   
` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00051da0
~ld
`

I use ESP07 with RFM69 and Standard Sketch. Compiled the code with Arduino IDE as well as with VS code, always with the same result. 
I disconnected RFM69 and checked the supply voltage. Have even used other calculator.
Have also tested ESP12e. 
Erase Flash I have also already done. 
However, when I run other code on the ESP, it runs without problems.
Erase Flash I have also already done. 
I just don't understand it.

`#define MY_DEBUG
#define MY_BAUD_RATE 115200
#define MY_RADIO_RFM69
#define MY_IS_RFM69HW

#define MY_GATEWAY_ESP8266

#define MY_WIFI_SSID "my ssid"
#define MY_WIFI_PASSWORD "my pw"

#define MY_HOSTNAME "MySensors-GW"
#define MY_PORT 5003
#define MY_GATEWAY_MAX_CLIENTS 3
#define MY_INCLUSION_MODE_FEATURE

// Enable Inclusion mode button on gateway
#define MY_INCLUSION_BUTTON_FEATURE
// Set inclusion mode duration (in seconds)
#define MY_INCLUSION_MODE_DURATION 60
// Digital pin used for inclusion mode button
#define MY_INCLUSION_MODE_BUTTON_PIN  3

#include <MySensors.h>

void setup()
{
}

void presentation()
{
}

void loop()
{
}
`

Zwer2k, 2021-08-10:
I found out that the error was due to too new version of espressif8266. I had version 3.1.0 installed, but MySensors seems to work only with versions < 3.0.0. Otherwise the error comes up. Unfortunately I did not find any hint about this anywhere.
Please add support vor new version.

mfalkvidd, 2021-08-11:
Crossposted in https://forum.mysensors.org/topic/11767/can-t-get-esp8266-gateway-to-work-anymore where answers are available.

closed 2021-08-11 by Zwer2k