mysensors/MySensors #1509: Error compiling for 3.x.x ESP8266 board when use both Wifi Manager and MySensor.

dungdao191299, 2021-09-21:
My code arduino for GatewayESP8266MQTTClient get error when compiling for 3.x.x ESP8266 Board. In my code, i used ESP8266WebServer lib for WifiManager Lib.
[image]

> This is my error:
> `
> In file included from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/StreamDev.h:25,
>                  from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:31,
>                  from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
>                  from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:62:
> C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/StreamString.h:54:50: error: macro "max" requires 2 arguments, but only 1 given
>    54 |         return std::numeric_limits<int16_t>::max();
>       |                                                  ^
> In file included from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:60:
> C:\Users\ADMIN\Documents\Arduino\libraries\MySensors/MySensors.h:98: note: macro "max" defined here
>    98 | #define max(a,b) ((a)>(b)?(a):(b)) //!< max
>       | 
> In file included from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:31,
>                  from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
>                  from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:62:
> C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/StreamDev.h:51:50: error: macro "max" requires 2 arguments, but only 1 given
>    51 |         return std::numeric_limits<int16_t>::max();
>       |                                                  ^
> In file included from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:60:
> C:\Users\ADMIN\Documents\Arduino\libraries\MySensors/MySensors.h:98: note: macro "max" defined here
>    98 | #define max(a,b) ((a)>(b)?(a):(b)) //!< max
>       | 
> In file included from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:31,
>                  from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
>                  from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:62:
> C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/StreamDev.h:118:50: error: macro "max" requires 2 arguments, but only 1 given
>   118 |         return std::numeric_limits<int16_t>::max();
>       |                                                  ^
> In file included from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:60:
> C:\Users\ADMIN\Documents\Arduino\libraries\MySensors/MySensors.h:98: note: macro "max" defined here
>    98 | #define max(a,b) ((a)>(b)?(a):(b)) //!< max
>       | 
> In file included from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:31,
>                  from C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
>                  from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:62:
> C:\Users\ADMIN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/StreamDev.h:145:50: error: macro "max" requires 2 arguments, but only 1 given
>   145 |         return std::numeric_limits<int16_t>::max();
>       |                                                  ^
> In file included from C:\Users\ADMIN\Desktop\20210921_gateway_new_topic\20210921_gateway_new_topic.ino:60:
> C:\Users\ADMIN\Documents\Arduino\libraries\MySensors/MySensors.h:98: note: macro "max" defined here
>    98 | #define max(a,b) ((a)>(b)?(a):(b)) //!< max
>       |`
[image]
Can Anyone help me?

dungdao191299, 2021-09-21:
I found a ưay to hack it. I move `#include <MySensors.h>` to the end of them. Like as:
[image]
My code was compiled and working. But do anyone know the reason? Sorry for my ignorance, i am a newbie

closed 2021-09-21 by dungdao191299

virtual-maker, 2021-09-26:
@dungdao191299 
This is because MySensors lib defines a macro named `max()`, see: [mysensors lib code](https://github.com/mysensors/MySensors/blob/aa76d266159d87cd460597657ac7f669a706fad5/MySensors.h#L98)

This max-macro brakes in multiple ESP8266 libraries code lines where max() is used in an different context, e.g. as a method of a class, or in your case the full qualified library function `std::numeric_limits<int16_t>::max()`, see: [esp lib code](https://github.com/esp8266/Arduino/blob/64e87f1149a0badf9438b29306d924db75eebf82/cores/esp8266/StreamString.h#L54)

Basically this is a bad design decision and should be changed in MySensors library.
The ESP library uses a macro for max which is called max with a leading underscore: `_max()`
This macro should be used in the MySensors code to stay compatible with the multiple ESP8266 libraries.

It would be perfect helpful if you could create an additional issue for this. 
Thank you