mysensors/MySensors #1514: Controller configuration info (isMetric) not shared in passive mode (MY_PASSIVE_NODE)

chey, 2021-12-01:
When passive mode is enabled in your sketch, the controller configuration isn't shared with the node.

In other words,
`getControllerConfig().isMetric` doesn't provide anything when using `#define MY_PASSIVE_NODE`.

I'll leave it up to the developers if you want this to be a bug or a change in documentation. If a change in the documentation, i would suggest somewhere on [this page](https://www.mysensors.org/download/sensor_api_20) at least.

virtual-maker, 2021-12-06:
Hello chey,

unfortunately I'm not able to reproduce your issue. With my passive node (according to the PassiveNode.ino example)
I get:
```
 __  __       ____
|  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
| |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
| |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
|_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
        |___/                      2.4.0-alpha

221 MCO:BGN:INIT NODE,CP=RUNPW---,FQ=NA,REL=0,VER=2.4.0-alpha
224 MCO:BGN:BFR
2374 TSM:INIT
2376 TSF:WUR:MS=0
3460 TSM:INIT:TSP OK
3461 TSM:INIT:TSP PSM
3461 TSM:INIT:STATID=18
3462 TSF:SID:OK,ID=18
3463 TSM:FPAR
3463 TSM:FPAR:STATP=255
3488 TSM:ID
3502 TSM:ID:OK
3516 TSM:UPL:DISABLED
3517 TSM:READY:ID=18,PAR=255,DIS=1
3518 ?TSF:MSG:SEND,18-18-255-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
3546 TSF:MSG:READ,0-63-18,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
3548 ?TSF:MSG:SEND,18-18-255-0,s=255,c=0,t=17,pt=0,l=11,sg=0,ft=0,st=OK:2.4.0-alpha
3554 ?TSF:MSG:SEND,18-18-255-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:255
3636 TSF:MSG:READ,0-63-18,s=255,c=3,t=6,pt=0,l=1,sg=0:M
8416 ?TSF:MSG:SEND,18-18-255-0,s=255,c=3,t=11,pt=0,l=12,sg=0,ft=0,st=OK:Passive Node
8422 ?TSF:MSG:SEND,18-18-255-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
8429 ?TSF:MSG:SEND,18-18-255-0,s=0,c=0,t=6,pt=0,l=0,sg=0,ft=0,st=OK:
8432 MCO:REG:NOT NEEDED
8458 MCO:BGN:STP
Is Metric
9587 MCO:BGN:INIT OK,TSP=1
9614 ?TSF:MSG:SEND,18-18-255-0,s=0,c=1,t=0,pt=7,l=5,sg=0,ft=0,st=OK:26,10
```
My setup() code is:
```
void setup()
{
	auto result = getControllerConfig();
	if (result.isMetric)
		Serial.println("Is Metric");
	else
		Serial.println("Is Imperial");
}
```
At 3461 you see transport is set to passive mode.
At 3636 the passive node gets Metric info from gateway and this is also print out at setup() after 8458.
I'm using HomeAssistant with Metric setting.

Are you sure that your gateway is connected with your controller and controller is set to Metric?

chey, 2021-12-07:
> Are you sure that your gateway is connected with your controller and controller is set to Metric?

Yes. I am pulling logs off the gateway anyway so this is what I'm getting in passive mode.

```
Dec 07 08:55:21 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
Dec 07 08:55:21 DEBUG SGN:PRE:SGN NREQ,FROM=192
Dec 07 08:55:21 DEBUG SGN:PRE:WHI NREQ,FROM=192
Dec 07 08:55:21 DEBUG SGN:PRE:SGN NREQ,TO=192
Dec 07 08:55:21 DEBUG SGN:PRE:WHI NREQ,TO=192
Dec 07 08:55:21 DEBUG SGN:SGN:NREQ=192
Dec 07 08:55:21 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
Dec 07 08:55:21 DEBUG SGN:PRE:XMT,TO=192
Dec 07 08:55:21 DEBUG TSF:MSG:READ,192-192-0,s=255,c=0,t=17,pt=0,l=11,sg=0:2.4.0-alpha
Dec 07 08:55:23 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=11,pt=0,l=18,sg=0:DistanceSensorTEST
Dec 07 08:55:23 DEBUG TSF:MSG:READ,192-192-0,s=1,c=0,t=15,pt=0,l=0,sg=0:
Dec 07 08:55:23 DEBUG TSF:MSG:READ,192-192-0,s=1,c=1,t=13,pt=2,l=2,sg=0:3
```

When not in passive mode the config is shared with the node.

```
Dec 07 09:00:41 DEBUG TSF:MSG:READ,192-192-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
Dec 07 09:00:41 DEBUG TSF:MSG:BC
Dec 07 09:00:41 DEBUG TSF:MSG:FPAR REQ,ID=192
Dec 07 09:00:41 DEBUG TSF:PNG:SEND,TO=0
Dec 07 09:00:41 DEBUG TSF:CKU:OK
Dec 07 09:00:41 DEBUG TSF:MSG:GWL OK
Dec 07 09:00:42 DEBUG SGN:SGN:NREQ=192
Dec 07 09:00:42 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
Dec 07 09:00:44 DEBUG TSF:MSG:PINGED,ID=192,HP=1
Dec 07 09:00:44 DEBUG SGN:SGN:NREQ=192
Dec 07 09:00:44 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=OK:1
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
Dec 07 09:00:44 DEBUG SGN:PRE:SGN NREQ,FROM=192
Dec 07 09:00:44 DEBUG SGN:PRE:WHI NREQ,FROM=192
Dec 07 09:00:44 DEBUG SGN:PRE:SGN NREQ,TO=192
Dec 07 09:00:44 DEBUG SGN:PRE:WHI NREQ,TO=192
Dec 07 09:00:44 DEBUG SGN:SGN:NREQ=192
Dec 07 09:00:44 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
Dec 07 09:00:44 DEBUG SGN:PRE:XMT,TO=192
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=0,t=17,pt=0,l=11,sg=0:2.4.0-alpha
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=6,pt=1,l=1,sg=0:0
Dec 07 09:00:44 DEBUG SGN:SGN:NREQ=192
Dec 07 09:00:44 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=OK:I
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=11,pt=0,l=18,sg=0:DistanceSensorTEST
Dec 07 09:00:44 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=12,pt=0,l=3,sg=0:1.0
Dec 07 09:00:45 DEBUG TSF:MSG:READ,192-192-0,s=1,c=0,t=15,pt=0,l=0,sg=0:
Dec 07 09:00:45 DEBUG TSF:MSG:READ,192-192-0,s=255,c=3,t=26,pt=1,l=1,sg=0:2
Dec 07 09:00:45 DEBUG SGN:SGN:NREQ=192
Dec 07 09:00:45 DEBUG TSF:MSG:SEND,0-0-192-192,s=255,c=3,t=27,pt=1,l=1,sg=0,ft=0,st=OK:1
Dec 07 09:00:45 DEBUG TSF:MSG:READ,192-192-0,s=1,c=1,t=13,pt=2,l=2,sg=0:3
```

virtual-maker, 2021-12-07:
@chey Yes, I see. In passive mode the config request from node to controller does not reach the gateway. 

Unfortunately we have no MY_DEBUG log from your passive node. But there should be a send request from node while presentation like in my log at 3554. Never less the node is passive or not, it always sends the config request, see in [code](https://github.com/mysensors/MySensors/blob/aa76d266159d87cd460597657ac7f669a706fad5/core/MySensorsCore.cpp#L255).

I assume this message is lost because of non perfect communication conditions between node and gateway. Pay attention that in passive mode (e.g. with NRF24 radios) each telegram is sent with NOACK. This means that the telegram is only sent once. With ACK (if no passive mode is used), the telegram is sent again up to 15 times if the receiver does not send back an acknowledgement.

chey, 2021-12-21:
Here is the debug output from a node using your example code.

```
17 MCO:BGN:INIT NODE,CP=NLNPA---,FQ=16,REL=0,VER=2.4.0-alpha
27 TSM:INIT
28 TSF:WUR:MS=0
40 TSM:INIT:TSP OK
41 TSM:INIT:TSP PSM
44 TSM:INIT:STATID=192
46 TSF:SID:OK,ID=192
48 TSM:FPAR
49 TSM:FPAR:STATP=255
51 TSM:ID
52 TSM:ID:OK
53 TSM:UPL:DISABLED
55 TSM:READY:ID=192,PAR=255,DIS=1
106 ?TSF:MSG:SEND,192-192-255-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
210 TSF:MSG:READ,0-0-192,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
280 ?TSF:MSG:SEND,192-192-255-0,s=255,c=0,t=17,pt=0,l=11,sg=0,ft=0,st=OK:2.4.0-alpha
337 ?TSF:MSG:SEND,192-192-255-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:255
2344 MCO:REG:NOT NEEDED
2347 MCO:BGN:STP
Is Metric
2348 MCO:BGN:INIT OK,TSP=1
60039 TSF:MSG:READ,22-22-0,s=1,c=1,t=0,pt=2,l=2,sg=0:50
60044 !TSF:MSG:REL MSG,NREP
```

I am also using HomeAssistant. It should be outputting `Is Imperial` but it's not.

*If I don't use passive mode it outputs `Is Imperial` like it should.*

virtual-maker, 2022-01-15:
Yes, at 337 your passive node sends the request for configuration (to get metric/imperial). 
But this telegram not reaches the gateway according your log from above. 
Then your node waits 2 seconds for configuration response but there is no response from gateway. 
So the node continues at 2344 and uses default config (which is metric). 

So for me this is a communication fault between passive node and gateway according my explanation from above.

mfalkvidd, 2022-09-04:
From the [documentation](https://www.mysensors.org/apidocs-beta/group__RoutingNodeSettingGrpPub.html#gacb52a7872ec805006f0f49f05c1e4fb2):
> #define MY_PASSIVE_NODE
> If enabled, the node operates fully autonomously, i.e. messages are sent without ACKing.

Since no acknowledgement is used, the node will be unable to detect lost messages and will therefore not make another request.

closed 2022-09-04 by mfalkvidd