mysensors/MySensors #1516: Need a better solution not to include PJON lib into MySensors repo

virtual-maker, 2022-02-09:
> @mfalkvidd For my understanding from Mr. Jenkins output the Toll gate fails because of a warning in the new PJON stuff, e.g. in line PJONDefines.h:207:8. May be the included PJON library does not follow the MySensors coding style guidelines?
> 
> BTW I personally dislike the idea to include this big external PJON library into the MySensors lib repo. May be an other solution by use of an external library reference would be better. But I have no idea how to do this :-(

The Arduino IDE does not handle dependencies (at least not in an useful way). This has caused a lot of problems for MySensors users. The least bad way we’ve found so far is to include all code in the same repo. But that also means that we in effect fork every dependency, which sucks.

I’m not sure how we should handle the pjon case. I’ll think about it and see if there is something I think could work better. Ideas and suggestions are very welcome.

_Originally posted by @mfalkvidd in https://github.com/mysensors/MySensors/issues/1515#issuecomment-1033165291_

mfalkvidd, 2022-02-10:
Related: https://github.com/mysensors/MySensorsArduinoExamples/issues/52

mfalkvidd, 2022-08-08:
[__has_include()](https://gcc.gnu.org/onlinedocs/cpp/_005f_005fhas_005finclude.html) might be useful, to display a user-friendly message with instruction on which library to install and how to install it.

But we still wouldn't know which version the included library is (unless the library itself has some define that determines the version) which makes it hard to ensure compatibility. Whenever the upstream library has breaking changes, MySensors users would be left clueless.

The same issue applies to TinyGSM, https://forum.mysensors.org/topic/12003/gsm-mqtt-gateway-can-we-update-the-tinygsm-library-used