mysensors/MySensors #1518: RPI mqtt gateway not working with user/password authentication

piotrraczynski, 2022-03-27:
I'm unable to connect to mqtt server (based on mosquitto 1.5.7) with user/password authentication. Passwordless connection works fine. At the same time I'm able to connect to the same broker with mosquitto_sub client. my gateway config:

`./configure --my-transport=rfm69 --my-rfm69-frequency=868 --my-is-rfm69hw --my-gateway=mqtt --my-controller-ip-address=192.168.1.45 --my-mqtt-publish-topic-prefix=mysensors-out --my-mqtt-subscribe-topic-prefix=mysensors-in --my-mqtt-client-id=mosquitto --my-leds-err-pin=12 --my-leds-rx-pin=16 --my-leds-tx-pin=18 --extra-cxxflags="-DMY_RFM69_NETWORKID=\(100\)" --my-mqtt-username=mosquitto --my-mqtt-password=123456`

debug output:

`Mar 27 21:07:17 INFO  Starting gateway...
Mar 27 21:07:17 INFO  Protocol version - 2.4.0-alpha
Mar 27 21:07:17 DEBUG MCO:BGN:INIT GW,CP=RPNGL---,FQ=NA,REL=0,VER=2.4.0-alpha
Mar 27 21:07:17 DEBUG TSF:LRT:OK
Mar 27 21:07:17 DEBUG TSM:INIT
Mar 27 21:07:17 DEBUG TSF:WUR:MS=0
Mar 27 21:07:17 DEBUG TSM:INIT:TSP OK
Mar 27 21:07:17 DEBUG TSM:INIT:GW MODE
Mar 27 21:07:17 DEBUG TSM:READY:ID=0,PAR=0,DIS=0
Mar 27 21:07:17 DEBUG MCO:REG:NOT NEEDED
Mar 27 21:07:17 DEBUG MCO:BGN:STP
Mar 27 21:07:17 DEBUG MCO:BGN:INIT OK,TSP=1
Mar 27 21:07:17 DEBUG GWT:RMQ:CONNECTING...
Mar 27 21:07:17 DEBUG connected to 192.168.1.45
Mar 27 21:07:18 DEBUG !GWT:RMQ:FAIL
Mar 27 21:07:18 DEBUG TSM:READY:NWD REQ
Mar 27 21:07:18 DEBUG ?TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
Mar 27 21:07:18 DEBUG GWT:RMQ:CONNECTING...
Mar 27 21:07:18 DEBUG connected to 192.168.1.45
Mar 27 21:07:19 DEBUG !GWT:RMQ:FAIL
Mar 27 21:07:19 DEBUG GWT:RMQ:CONNECTING...
Mar 27 21:07:19 DEBUG connected to 192.168.1.45
Mar 27 21:07:20 DEBUG !GWT:RMQ:FAIL
Mar 27 21:07:20 DEBUG GWT:RMQ:CONNECTING...
Mar 27 21:07:20 DEBUG connected to 192.168.1.45`

mosquitto log:
`1648415219: New connection from 192.168.1.105 on port 1883.
1648415219: Socket error on client <unknown>, disconnecting.
1648415220: New connection from 192.168.1.105 on port 1883.
1648415220: Socket error on client <unknown>, disconnecting.
1648415221: New connection from 192.168.1.105 on port 1883.
1648415221: Socket error on client <unknown>, disconnecting.
1648415222: New connection from 192.168.1.105 on port 1883.
1648415222: Socket error on client <unknown>, disconnecting.
1648415223: New connection from 192.168.1.105 on port 1883.
1648415223: Socket error on client <unknown>, disconnecting.
1648415224: New connection from 192.168.1.105 on port 1883.
1648415224: Socket error on client <unknown>, disconnecting.
1648415225: New connection from 192.168.1.105 on port 1883.
1648415225: Socket error on client <unknown>, disconnecting.
1648415226: New connection from 192.168.1.105 on port 1883.
1648415226: Socket error on client <unknown>, disconnecting.
1648415227: New connection from 192.168.1.105 on port 1883.
1648415227: Socket error on client <unknown>, disconnecting.
1648415228: New connection from 192.168.1.105 on port 1883.
1648415228: Socket error on client <unknown>, disconnecting.
1648415229: New connection from 192.168.1.105 on port 1883.
1648415229: Socket error on client <unknown>, disconnecting.
1648415230: New connection from 192.168.1.105 on port 1883.
1648415230: Socket error on client <unknown>, disconnecting.
1648415231: New connection from 192.168.1.105 on port 1883.
1648415231: Socket error on client <unknown>, disconnecting.
1648415232: New connection from 192.168.1.105 on port 1883.
1648415232: Socket error on client <unknown>, disconnecting.
1648415233: New connection from 192.168.1.105 on port 1883.
1648415233: Socket error on client <unknown>, disconnecting.
1648415234: New connection from 192.168.1.105 on port 1883.
1648415234: Socket error on client <unknown>, disconnecting.
1648415235: New connection from 192.168.1.105 on port 1883.
1648415235: Socket error on client <unknown>, disconnecting.
1648415237: New connection from 192.168.1.105 on port 1883.
1648415237: Socket error on client <unknown>, disconnecting.
1648415238: New connection from 192.168.1.105 on port 1883.
1648415238: Socket error on client <unknown>, disconnecting.
1648415239: New connection from 192.168.1.105 on port 1883.`

piotrraczynski, 2022-03-28:
Also tried 2.0.14 mosquitto from source. On that version, with some debugging logs I got (from the broker):

1648458618: New connection from 192.168.1.105:57522 on port 1883.
1648458618: Sending CONNACK to mosquitto (0, 5)
1648458618: Client mosquitto disconnected, not authorised.
1648458618: New connection from 192.168.1.105:57524 on port 1883.
1648458618: Sending CONNACK to mosquitto (0, 5)
1648458618: Client mosquitto disconnected, not authorised.

the password I provide is correct.

mfalkvidd, 2022-04-01:
Is the password as simple as 123456, or does it contain any special characters that might become mangled by the shell or the configure script?

piotrraczynski, 2022-04-01:
I was not able to get any password working. I started with more complicated
one but then settled with 123456. I also printed username and password from
mysgw itself as debug strings. They do not look mangled in any way.

Anyway, at least temporary I started to use mqtt with no password but
actually setup my external gateways in zerotier network.

mfalkvidd, 2022-04-02:
@piotrraczynski I think I found the problem. The parameter to ./configure is `--my-mqtt-user`, not `--my-mqtt-username`

mfalkvidd, 2022-04-02:
The configure command actually outputs this when inputting my-mqtt-username:
```
[WARNING] Unknown option detected:--my-mqtt-username=mosquitto, ignored
```
But the warning is very easy to miss, I missed it myself.

mfalkvidd, 2022-04-10:
@piotrraczynski have you had a chance to verify the suggested solution?

mfalkvidd, 2022-08-28:
Closing due to inactivity and what (probably) was a configuration error. Feel free to reopen.

closed 2022-08-28 by mfalkvidd