mysensors/MySensors #1523: Watchdog timer overflow error for big timeout

mrespin, 2022-06-08:
In file hal/architecture/NRF5/drivers/wdt.h line 80:
NRF_WDT->CRV = (32768*timeout)/1000; \
Setting large timeout value generating overflow error.
If changed to
NRF_WDT->CRV = (uint32_t)(32.768*timeout); \
there is no overflow error anymore. Not sure if (uint32_t) cast is needed.

oxyshad, 2022-06-08:
You can define the type also in wdt_enable call. 
Example with timer of 60 minutes
wdt_enable(60*60*1000**UL**)

mrespin, 2022-06-09:
Yes, that is the problem, as wdt_enable expands to:
#define wdt_enable(timeout) \
	NRF_WDT->CONFIG = NRF_WDT->CONFIG = (WDT_CONFIG_HALT_Pause << WDT_CONFIG_HALT_Pos) | ( WDT_CONFIG_SLEEP_Run << WDT_CONFIG_SLEEP_Pos); \
	**NRF_WDT->CRV = (32768 * timeout)/1000; \**
	NRF_WDT->RREN |= WDT_RREN_RR0_Msk;  \
	NRF_WDT->TASKS_START = 1
in hal/architecture/NRF5/drivers/wdt.h

mrespin, 2022-07-17:
I hope that will help.
src\main.cpp:251:34: warning: integer overflow in expression [-Woverflow]
#define timeout 5 * 60 * 1000  
.pio\...\MySensors/hal/architecture/NRF5/drivers/wdt.h:80:24: note: in definition of macro 'wdt_enable'
NRF_WDT->CRV = (32768*timeout)/1000; \
                                        ^~~~~~~
Working fix: 
NRF_WDT->CRV = (32.768*timeout); \