mysensors/MySensors #1536: NRF24 gateway depends on the deprecated GPIO Linux interface

aledomu, 2022-12-05:
The gateway opens `/sys/class/gpio` when accessing GPIO ports. In distros where they follow upstream standards this makes the app unable to work. This interface was deprecated in Linux 4.8.

mfalkvidd, 2022-12-05:
@aledomu du you happen to know what the recommended replacement is?

mfalkvidd, 2022-12-05:
Looks like the spi_gpio kernel driver might be the way to go:
> - spi_gpio: drivers/spi/spi-gpio.c is used to drive an SPI bus (variable number
  of wires, at least SCK and optionally MISO, MOSI and chip select lines) using
  GPIO hammering (bitbang). It will appear as any other SPI bus on the system
  and makes it possible to connect drivers for SPI devices on the bus like
  any other SPI bus driver. For example any MMC/SD card can then be connected
  to this SPI by using the mmc_spi host from the MMC/SD card subsystem.

But I have no idea if this driver is available (and usable with normal user privileges) by default, nor how much work would be required to rewrite the MySensors code to use this driver.

aledomu, 2022-12-08:
The current recommended interface is exposed through `/dev/gpiochip*` character nodes.